mani-skill/ManiSkill #8: [Minor] Default observation in some soft body envs seems to be `pointcloud` (leads to different expected behavior).

DanielTakeshi, 2022-08-09:
The current README states for `tools/replay_trajectory.py`:

> We recommend using our script only for converting actions into different control modes and not record any observation information (i.e. without passing `--obs-mode`). The reason is that (1) some observation modes, e.g. point cloud, can take much space without any post-processing, e.g., point cloud downsampling; in addition, the state mode for soft-body environments also has a similar issue, since the states of those environments are particles. (2) Some algorithms (e.g. GAIL) require custom keys stored in the demonstration files, e.g. next-observation.

However it seems like for 3 of the soft body envs, after loading the demonstrations, the `tools/replay_trajectory.py` will save using the point cloud type even without `--obs-mode` explicitly passed on the command line. Thus, by default the code will do the "non recommended" behavior as per the README. 

Running this command:

```
python tools/replay_trajectory.py --traj-path demos/soft_body_envs/Excavate-v0/trajectory.h5   --save-traj --target-control-mode pd_joint_delta_pos --num-procs 10
```

Will create:

```
demos/soft_body_envs/Excavate-v0:
total 21G
-rw-rw-r-- 1 seita seita 219M Aug  4 11:28 trajectory.h5
-rw-rw-r-- 1 seita seita  57K Aug  4 11:28 trajectory.json
-rw-rw-r-- 1 seita seita  21G Aug  9 10:59 trajectory.pointcloud.pd_joint_delta_pos.h5
-rw-rw-r-- 1 seita seita  60K Aug  9 10:59 trajectory.pointcloud.pd_joint_delta_pos.json
```

Whereas for another env like `PegInsertionSide-v0` I will see:

```
demos/rigid_body_envs/PegInsertionSide-v0:
total 97M
-rw-rw-r-- 1 seita seita  49M Aug  4 11:27 trajectory.h5
-rw-rw-r-- 1 seita seita 398K Aug  4 11:27 trajectory.json
-rw-rw-r-- 1 seita seita  48M Aug  9 10:23 trajectory.none.pd_joint_delta_pos.h5
-rw-rw-r-- 1 seita seita 403K Aug  9 10:23 trajectory.none.pd_joint_delta_pos.json
```

So here we have `trajectory.none` as the prefix instead of `trajectory.pointcloud` and the file size is much smaller for the former.

The reason is I think that the `trajectory.json` files we load has `pointcloud` as the `obs_mode` for soft body envs, for example:

```
(mani_skill2) seita@takeshi:~/ManiSkill2 (main)$ head -n 20 demos/soft_body_envs/Excavate-v0/trajectory.json 
{
  "env_info": {
    "env_id": "Excavate-v0",
    "max_episode_steps": 250,
    "env_kwargs": {
      "obs_mode": "pointcloud",
      "reward_mode": "dense",
      "control_mode": "pd_joint_pos"
    }
  },
  "commit_info": {
    "commit_id": "ef73ec44f5f0f22c9b2f71e682fbdca0c7d377fa",
    "branch": "yihe-dev"
  },
  "episodes": [
    {
      "episode_id": 0,
      "episode_seed": 1,
      "reset_kwargs": {
        "seed": 1,
(mani_skill2) seita@takeshi:~/ManiSkill2 (main)$ 
```

The same is true with `Fill-v0` and `Hang-v0` though interestingly not `Pour-v0` (which uses `particle` for some reason). The rigid body envs have `none` as the observation mode.

To be clear, this doesn't affect the _correctness_ of the code, though it might be easier (possibly for future users?) if they can use the "none" observation type to save on disk space. Or maybe a possible remark to this in the README could help clarify?

Thanks for the code and environments. :)

xiqiangliu, 2022-08-09:
Yes, the default behavior of `replay_trajectory.py` is replay under the `obs_mode` of the input trajectories. The reason for such design is that some environments could potentially not have a `none` observation mode and replaying under `none` will lead to an error. We will update the README soon to point out this behavior.

Thank you for bringing up this suggestion!

xuanlinli17, 2022-08-09:
README pull request submitted.

closed 2022-08-09 by xuanlinli17

DanielTakeshi, 2022-08-13:
@xuanlinli17 Sorry for bumping a closed report (I can make a new one if you want), but how do you suggest dealing with `Pour-v0` which has the `particles` type? If I do something like:

```
python tools/replay_trajectory.py --traj-path /data/seita/maniskill2/demos/soft_body_envs/Pour-v0/trajectory.h5   --save-traj  --target-control-mode pd_joint_delta_pos --num-procs 5
```

Then I will get:

```
Traceback (most recent call last):
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/pool.py", line 51, in starmapstar
    return list(itertools.starmap(args[0], args[1]))
  File "/home/seita/ManiSkill2/tools/replay_trajectory.py", line 266, in _main
    ori_env = gym.make(env_id, **ori_env_kwargs)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 76, in make
    env = cls(**_kwargs)
  File "/home/seita/ManiSkill2/mani_skill2/envs/mpm/pour_env.py", line 98, in __init__
    super().__init__(*args, **kwargs)
  File "/home/seita/ManiSkill2/mani_skill2/envs/mpm/base_env.py", line 92, in __init__
    super().__init__(*args, **kwargs)
  File "/home/seita/ManiSkill2/mani_skill2/envs/sapien_env.py", line 86, in __init__
    raise NotImplementedError("Unsupported obs mode: {}".format(obs_mode))
NotImplementedError: Unsupported obs mode: particles
```

This is the only remaining environment where I'm not sure about the intended behavior as per the README. The other 7 available envs all seem to be working as intended with `tools/replay_trajectory.py` as per the current README.

xuanlinli17, 2022-08-13:
We have updated our readme (i.e. `passing --obs-mode=none`).

DanielTakeshi, 2022-08-14:
Ah! Apologies, I was looking at an older commit. I have tried the command from the existing README and with `--obs-mode none` (or `--obs-mode=none`, `--obs-mode=None`) I get:

```
(mani_skill2) seita@takeshi:~/ManiSkill2 (main)$ python tools/replay_trajectory.py --traj-path /data/seita/maniskill2/demos/soft_body_envs/Pour-v0/trajectory.h5  --save-traj --target-control-mode pd_joint_delta_pos --obs-mode none --num-procs 5
                      Warp initialized:
   Version: 0.3.1ep/s]
   CUDA device: NVIDIA GeForce RTX 3090
   Kernel cache: /home/seita/.cache/warp/0.3.1
Warp initialized:ep/s]
   Version: 0.3.1
   CUDA device: NVIDIA GeForce RTX 3090
   Kernel cache: /home/seita/.cache/warp/0.3.1
Warp initialized:
   Version: 0.3.1
   CUDA device: NVIDIA GeForce RTX 3090
   Kernel cache: /home/seita/.cache/warp/0.3.1
Warp initialized:
   Version: 0.3.1
   CUDA device: NVIDIA GeForce RTX 3090
   Kernel cache: /home/seita/.cache/warp/0.3.1
Warp initialized:
   Version: 0.3.1
   CUDA device: NVIDIA GeForce RTX 3090
   Kernel cache: /home/seita/.cache/warp/0.3.1
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/pool.py", line 51, in starmapstar
    return list(itertools.starmap(args[0], args[1]))
  File "/home/seita/ManiSkill2/tools/replay_trajectory.py", line 266, in _main
    ori_env = gym.make(env_id, **ori_env_kwargs)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 76, in make
    env = cls(**_kwargs)
  File "/home/seita/ManiSkill2/mani_skill2/envs/mpm/pour_env.py", line 98, in __init__
    super().__init__(*args, **kwargs)
  File "/home/seita/ManiSkill2/mani_skill2/envs/mpm/base_env.py", line 92, in __init__
    super().__init__(*args, **kwargs)
  File "/home/seita/ManiSkill2/mani_skill2/envs/sapien_env.py", line 86, in __init__
    raise NotImplementedError("Unsupported obs mode: {}".format(obs_mode))
NotImplementedError: Unsupported obs mode: particles
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "tools/replay_trajectory.py", line 423, in <module>
    main()
  File "tools/replay_trajectory.py", line 404, in main
    res = pool.starmap(_main, proc_args)
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/pool.py", line 372, in starmap
    return self._map_async(func, iterable, starmapstar, chunksize).get()
  File "/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/pool.py", line 771, in get
    raise self._value
NotImplementedError: Unsupported obs mode: particles
/home/seita/miniconda3/envs/mani_skill2/lib/python3.8/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 5 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
(mani_skill2) seita@takeshi:~/ManiSkill2 (main)$ 
```

I also get the same with `--obs-mode=pointcloud` and `--obs-mode=particles`, and `--obs-mode=rgbd` into the above command. Am I using the command correctly? Note that I moved the `demos/` directory to a new directory on my hard disk drive in the command but that shouldn't have affected things.

xuanlinli17, 2022-08-14:
Could you redownload the demonstrations? We recently changed all raw soft body demonstrations to use `obs_mode=none`.

DanielTakeshi, 2022-08-14:
That seems to have fixed it! Thanks for the recommendation to re-download.
Though to clarify this is not going to change the nature of the demonstrations (they will still follow the same trajectory when you visualize them) but just changing some metadata?

xuanlinli17, 2022-08-14:
yes it is.