mani-skill/ManiSkill #18: More objects of partnet_mobility

ZhuFengdaaa, 2022-08-30:
In [this page](https://sapien.ucsd.edu/downloads), you released 10 thousand partet mobility objects. However, the `python tools/download.py --uid faucet` command can only download hundreds of them. The data format is different. Faucet task reads the `moblity_cvx.urdf` file while the released large-scale dataset provide only `mobility.urdf`. 

Could you please release the preprocess script or the processed data of other partnet mobility objects?

Jiayuan-Gu, 2022-08-30:
Hi, in order to keep hold-out assets private for the [ManiSkill 2022 challenge](https://sapien.ucsd.edu/challenges/maniskill/2022/), we will not release the preprocessing script. 

However, I can share the general idea about how to preprocess articulations.
1. Parse URDF (recommended tools: [bs4](https://www.crummy.com/software/BeautifulSoup/bs4/doc/), https://github.com/mmatl/urdfpy, pytransform3d)
2. For each link, merge all visual meshes, and do convex decomposition (V-HACD, [COACD](https://github.com/SarahWeiii/CoACD)) for the merged mesh to get the collision mesh of the link.
3. Save the collision mesh and add collision tag in URDF
4. Save URDF

Jiayuan-Gu, 2022-09-03:
The processed assets of [ManiSkill 2021](https://github.com/haosulab/ManiSkill) will be released in the next version.

closed 2022-09-03 by Jiayuan-Gu