mani-skill/ManiSkill #25: Rendering with kuafu throws segmentation randomly

Xingyu-Lin, 2022-09-13:
Hi,

After I update with the most recent commints on the kuafu renderer, I get segemtantion fault when running the kuafu renderer. Below is a minimal script to hopfully reproduce the issue I face. 

```
import gym
import numpy as np
from mani_skill2.utils.wrappers import RecordEpisode

def main():
    np.set_printoptions(suppress=True, precision=3)
    env = gym.make(
        'TurnFaucet-v0',
        obs_mode=None,
        reward_mode=None,
        control_mode='pd_ee_delta_pose',
        enable_kuafu=True)
    record_dir = './'
    env = RecordEpisode(env, record_dir, render_mode='cameras')

    frames = []
    for i in range(10):
        env.reset()
        while True:
            render_frame = env.render(mode='cameras')
            frames.append(render_frame)

            action = env.action_space.sample()
            obs, reward, done, info = env.step(action)
            if done:
                break
        from my_utils.visualization_utils import save_numpy_as_video
        save_numpy_as_video(np.array(frames), f'./test_data/test_{i}.mp4')

if __name__ == "__main__":
    main()
```

When I run the above script, I wil l get the following error. Sometimes it will throw a segmentation fault instead after a few trials.
```
[2022-09-13 00:21:16.995] [kuafu] [info] Camera is not yet usable due to uninitialized context!
[2022-09-13 00:21:16.995] [kuafu] [info] Offscreen mode enabled.
[2022-09-13 00:21:16.995] [kuafu] [warning] Denoiser ON! You must have an NVIDIA GPU with driver version > 470 installed.
2022-09-13 00:21:17,948 - mani_skill2 - WARNING - Only rgb is supported by KuafuRenderer.
trial 0
trial 1
Traceback (most recent call last):
  File "h5py/_objects.pyx", line 201, in h5py._objects.ObjectID.__dealloc__
RuntimeError: Can't decrement id ref count (file write failed: time = Tue Sep 13 00:21:46 2022
, filename = '20220913_002118.h5', file descriptor = 52, errno = 9, error message = 'Bad file descriptor', buf = 0x556c1261aae8, total write size = 64320, bytes this sub-write = 64320, bytes actually written = 18446744073709551615, offset = 18446744073709551615)
Exception ignored in: 'h5py._objects.ObjectID.__dealloc__'
Traceback (most recent call last):
  File "h5py/_objects.pyx", line 201, in h5py._objects.ObjectID.__dealloc__
RuntimeError: Can't decrement id ref count (file write failed: time = Tue Sep 13 00:21:46 2022
, filename = '20220913_002118.h5', file descriptor = 52, errno = 9, error message = 'Bad file descriptor', buf = 0x556c1261aae8, total write size = 64320, bytes this sub-write = 64320, bytes actually written = 18446744073709551615, offset = 18446744073709551615)

```

Jiayuan-Gu, 2022-09-14:
Thanks for providing us with a minimal example. However, I am not able to reproduce segfault given this script. KuafuRenderer is experimentally supported in ManiSkill2, and we hope the community can help us improve relevant features together.

After some quick checks, I think the error results from creating new scenes for different assets. If we only use one model, then the env will not create any new scene, which will not raise segfault.

Here is my modified minimal example:
```python
import gym
import numpy as np
from mani_skill2.utils.wrappers import RecordEpisode

def main():
    np.set_printoptions(suppress=True, precision=3)
    env = gym.make(
        "TurnFaucet-v0",
        obs_mode=None,
        model_ids=["5001", "5021"],
        reward_mode=None,
        control_mode="pd_ee_delta_pose",
        enable_kuafu=True,
    )
    record_dir = "./tmp/issue25"
    # env = RecordEpisode(env, record_dir, render_mode="cameras")

    frames = []
    for i in range(10):
        print("Episode", i)
        env.reset()
        for _ in range(10):
            render_frame = env.render(mode="cameras")
            frames.append(render_frame)

            action = env.action_space.sample()
            obs, reward, done, info = env.step(action)
            print("step", info["elapsed_steps"])
            if done:
                break

if __name__ == "__main__":
    main()
```

The workaround now can be:
1. using environments without asset change
2. only use one model each time (PickSingleYCB, PickSingleEGAD, TurnFaucet all support this functionality through initialization, and all environments also support passing model_ids when reset) 
3. If you want to replay trajectories, I suggest you first split trajectories by model ids according to the associated json file, and use a bash script to process each model.

@jetd1 Please help check how the KuafuRender's memory is managed for a new scene.

Xingyu-Lin, 2022-09-15:
Indeed, the issue comes from creating a new scene with a different model. Use one process for each model addressed the issue, although the solution does not seem ideal.

Xingyu-Lin, 2022-09-16:
A related issue: When the save_trajectory option is set, I got wierd error from h5py, as shown below. This error is gone if I comment out the rendering.

```
Traceback (most recent call last):
  File "examples/demo_manual_control.py", line 57, in <module>
    main()
  File "examples/demo_manual_control.py", line 49, in main
    render_traj(0)
  File "examples/demo_manual_control.py", line 43, in render_traj
    env.close()
  File "/home/xingyu/Projects/OpenWorldManipulation/mani_skill2/utils/wrappers/record.py", line 277, in close
    self._h5_file.close()
  File "/home/xingyu/software/miniconda3/envs/maniskill2/lib/python3.8/site-packages/h5py/_hl/files.py", line 552, in close
    self.id._close_open_objects(h5f.OBJ_LOCAL | h5f.OBJ_FILE)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 360, in h5py.h5f.FileID._close_open_objects
RuntimeError: Can't decrement id ref count (unable to extend file properly, errno = 9, error message = 'Bad file descriptor')
Segmentation fault (core dumped)
```

Minimal example to reproduce this issue:
```python
from mani_skill2.utils.wrappers import RecordEpisode
from multiprocessing import Pool
import multiprocessing as mp

def render_traj(i):
    print('render traj ', i)
    import gym
    import numpy as np
    np.set_printoptions(suppress=True, precision=3)
    model_ids = ['5001', '5021']
    model_id = model_ids[i % len(model_ids)]
    env = gym.make(
        'TurnFaucet-v0',
        obs_mode=None,
        reward_mode=None,
        control_mode='pd_ee_delta_pose',
        enable_kuafu=True)

    record_dir = '/tmp/issue25/'
    env = RecordEpisode(env,
                        record_dir,
                        render_mode='cameras',
                        save_trajectory=True,
                        save_on_reset=False,
                        save_video=True,
                        trajectory_name=str(i))
    env.reset(model_id=model_id)
    env.close()

def main():
    render_traj(0)

if __name__ == "__main__":
    mp.set_start_method('spawn')
    main()

```

jetd1, 2022-10-01:
This bug should be fixed with https://github.com/haosulab/SAPIEN/commit/31e12dd0f657bf211bf52432c657097dbeff078a
Please consider building SAPIEN from scratch if you need the fix immediately (https://github.com/haosulab/SAPIEN/blob/dev/docker_install_38_debug.sh).

closed 2022-10-01 by jetd1

Jiayuan-Gu, 2022-10-10:
https://github.com/haosulab/ManiSkill2/issues/25#issuecomment-1248769861 seems to be relevant to the accidental corruption of hdf5. However, the minimal example seems to be irrelevant and I can not reproduce it yet. I may comment on this issue if I figure out what happens.