mani-skill/ManiSkill #26: Rendering with kuafu of different size gives noisy image

Xingyu-Lin, 2022-09-16:
With the kuafu renderer, I tried to change the default camera size from 512 to 1024. This is done by changing this line: https://github.com/haosulab/ManiSkill2/blob/787439b2484190bcbbca2e2daa9aa2cea38816be/mani_skill2/envs/experimental/turn_faucet.py#L74

However, I got noisy rendered results. See the noisy pixels in the sky. Do you know why this might happen?

[image]

jetd1, 2022-09-19:
Thanks for this report. Can you reproduce the same artifacts every time you run the script? What happen if you change the viewing angle? Are these artifacts moving with the camera? Or are they taking up the same pixels?

Xingyu-Lin, 2022-09-20:
I have not tried different viewing angles. But if I use a different resolution, the location of the artifacts will change

Jiayuan-Gu, 2023-02-15:
Hi @Xingyu-Lin 

We have released `mani_skill2==0.4.0`. Fanbo has made efforts to integrate KuafuRenderer into VulkanRenderer, which is unified under `SapienRenderer`. More features and improvements are included in the new version. Please see the [release note](https://github.com/haosulab/ManiSkill2/releases/tag/v0.4.0) for details. Feel free to let us know what we can improve. Thank you.

Xingyu-Lin, 2023-02-15:
Thanks, the new features look amazing!

closed 2023-02-15 by Xingyu-Lin