mani-skill/ManiSkill #33: Mass for YCB models?

saneeeee, 2022-11-09:
Hello, 
I was wondering where mass for YCB models are stored, or are they ignored in simulation?
Thank you!

Jiayuan-Gu, 2022-11-10:
Hi, I specify the density for each YCB model in [this file](https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/assets/mani_skill2_ycb/info_pick_v0.json). The simulator will compute the mass on the fly given its collision mesh. You can acquire the mass by `env.obj.mass` for `PickSingleYCB-v0`. I roughly refer to the official document of YCB: https://people.eecs.berkeley.edu/~pabbeel/papers/2015-RAM-YCB-object-set.pdf.    We are considering using the official mass for each object in a later release. Feel free to let us know if you have any suggestions.

saneeeee, 2022-11-10:
Great, thank you! As a follow up, are there any benefits to or other specific reasons why storing density was preferred over mass?

May I also ask if the collision meshes for models were constructed by the Mani Skill team or acquired from somewhere else?

Thank you!

Jiayuan-Gu, 2022-11-10:
The reason is that the simulator usually needs to set mass and inertia for rigid objects (SAPIEN requires to specify both when building actors). I have not found inertia from the official YCB dataset. Besides, in order to guarantee that objects can be grasped by the Panda gripper, we need to scale some objects. Thus, a reasonable density can be more flexible than mass.

We use [CoACD](https://github.com/SarahWeiii/CoACD), an alternative to VHACD, to generate collision meshes. Collision meshes for all YCB objects are available elsewhere (e.g., AI Habitat), but they are not fully tested to be feasible for grasp.

saneeeee, 2022-11-10:
Sounds great, Nvidia IsaacGym has the option to calculate inertia directly from geometry. Which seems to work fine in my experience. 

However, it weirdly distorts the collision meshes that I get from Mani Skill, and so almost always needs VHACD, which sometimes results in degenerate collision meshes. I believe that's a problem on IsaacGym's side, since it does that with [vanilla YCB models ](https://www.ycbbenchmarks.com) too.

Thanks a lot for your input, best!

Jiayuan-Gu, 2022-11-15:
Our collision meshes have already been processed to be convex. The possible issue I can come up with is that I use trimesh to export those models and they do not contain the "convex_x" label in the obj, which usually are generated by V-HACD. I am not sure how Isaac Gym deals with it. We would appreciate it if you coult let us know what happens when you figure it out.

closed 2022-11-15 by Jiayuan-Gu