mani-skill/ManiSkill #35: gravity in pinch_env

hongkai-dai, 2022-11-15:
The mpm model gravity is (0, 0, -1) in the pinch environment https://github.com/haosulab/ManiSkill2/blob/0f72be61a866752609039b1e797403d10c29e42b/mani_skill2/envs/mpm/pinch_env.py#L59, may I ask why it is not set to -9.8 as in the original MPMModel https://github.com/haosulab/ManiSkill2/blob/0f72be61a866752609039b1e797403d10c29e42b/warp_maniskill/warp/sim/model.py#L357?

fbxiang, 2022-11-15:
This design is mainly for simulation stability. We find a stable simulation of the material with a normal gravity would require a much smaller time step and making the simulation much slower. We tune the gravity to be smaller similar to other frameworks like PlasticineLab.

hongkai-dai, 2022-11-15:
I see, thanks a lot for the prompt reply! That makes sense.

closed 2022-11-15 by hongkai-dai