mani-skill/ManiSkill #41: Install of ManiSkill2 fails

caiqi, 2022-11-29:
Thanks for open-sourcing the awesome work. I have trouble installing ManiSkill2 on a server and hope you can provide some help. Thanks. 

1. I am using a training platform so I can only run commands inside the docker container. The NVIDIA driver version is 470.57.02. I could not change the version of the NVIDIA driver due to platform limitations. There is no libGLX_nvidia.so file inside the docker. So I download the NVIDIA driver 470.57.02 from the official [website](https://www.nvidia.com/Download/driverResults.aspx/177145/en-us/) and extract all files from the driver file. I find libGLX_nvidia.so.470.57.02 and nvidia_icd.json and save them to /dev/shm/base folder. 
[image]
[image]

I also find the vulkan libs:
[image]

2. The command I run is: 
[image]

It seems there is something wrong with the render. Could you please give me some hints how to solve this problem?

Jiayuan-Gu, 2022-11-29:
Hi, could you first try the method described in #34 to verify your Vulkan installation? Besides, are you able to use a different docker image to verify whether it is an issue due to the docker configuration? We provide an example [docker file](https://github.com/haosulab/ManiSkill2/blob/main/docker/Dockerfile) and a built docker image (haosulab/maniskill2022-challenge:latest).

caiqi, 2022-11-30:
1. The information of nvidia-smi and vulkaninfo is : 
[image]
2. I follow the docker file to add libvulkan1 and mesa-utils into my current container, the problem still exist. 
3. I try to pull haosulab/maniskill2022-challenge:latest from the docker hub and find that there is also no libGLX_nvidia.so file inside the container, is this normal or I should use some specific command to start the container?

fbxiang, 2022-11-30:
Based on the information you provided, my suggestion is to fix the NVIDIA driver installation on the **host** machine. I have not seen a case where unpacking a downloaded NVIDIA driver could "fix" a faulty installation. I suggest only using ways supported by NVIDIA to install their driver (run file or through package manager) on the **host** machine.

`libGLX_nvidia.so` missing in the container is more or less expected. When you pass `--gpus all` to a docker command, the graphics driver, including all the `/usr/lib/**nvidia**.so` and other related files are mapped to the container. So if those files are missing in the host, you will not get them in the container, and manually copying them into the container probably will not help.

So I believe vulkaninfo is giving you the correct message that you have an incompatible driver on the host machine. The host driver essentially decides to disable graphics capability of your GPU.

caiqi, 2022-11-30:
Thanks for your timely reply! Currently, the host machine is a cloud cluster so I have to find another workaround.  "The host driver essentially decides to disable graphics capability of your GPU." Do you mean that the host driver can disable graphics capability so that I can normally use CUDA/PyTorch inside the container but could not use vulkan or sapien?

fbxiang, 2022-11-30:
Yes, libGLX implements many graphics features, including OpenGL and Vulkan. CUDA applications (e.g. pytorch) are not affected. Your host driver does not have graphics support and the only solution I know is to (maybe convince the admin and) do a full installation of the NVIDIA driver. If you find some other workaround, please do let us know, as it is an issue that happens quite often to people wanting to run graphics applications on cloud servers without graphics libraries.

closed 2022-12-01 by caiqi