mani-skill/ManiSkill #42: Support of granular object task

Kaiwen-Hong, 2022-12-08:
Thanks for providing so many interesting and challenging tasks for the rigid body and soft body! I'm wondering if the simulator could support testing granular objects, e.g. sand, coffee beans, or nuts. If so, what is the possible advantage and disadvantage of the ManiSkill2 simulator compared with others, like PyFlex, Isaac gym, or Tachi-MPM?

fbxiang, 2022-12-14:
First, Even though sand and coffee beans are both granular objects, they are typically simulated with completely different techniques due to particle size. 

Our simulator is similar to Taichi-MPM, which can do sand simulation with MPM (the sand material is implemented, but not actually used in the challenge). The major difference is that our simulator supports 2-way coupling and using a robot arm. I believe Taichi-MPM only does 1-way coupling and you need to implement the robot arm by yourself. So the Taichi code is simpler but also has less functionalities. PyFlex can also do some similar simulation but it also lacks 2-way coupling and robot support. The good thing about Flex is that it is faster, but at the cost of using a  material model not based on physics.

Big and irregular particles such as coffee beans and nuts are simulated as rigid body most of the time. You cannot really use MPM or Flex to do it easily. The only difference between ManiSkill and Isaac gym is that ManiSkill runs on CPU and Isaac gym runs on GPU (the simulation behavior should be almost identical).

Kaiwen-Hong, 2022-12-15:
Thanks for your reply! As far as I know, Isaac gym has two engines flex and PhysX. You have talked about some advantages/disadvantages of flex/pyflex. What about the one using PhysX engine?

fbxiang, 2022-12-15:
SAPIEN uses PhysX 4, and has a PhysX 5 branch. So on the physics side, ManiSkill is almost the same as Isaac gym with PhysX. It could be exactly the same with matching parameters.
We choose to simulate on CPU since the GPU solver of PhysX is not open source.
 The flex in Isaac gym is more complicated. It is actually very different from pyflex. It is an  upgraded version and supports robots. However, my understanding is that it is more closed down (there is no SDK now), so you are essentially forced to use Isaac and cannot wrap it into pyflex. It is probably better to ask them for more information as Isaac is also under development.

Kaiwen-Hong, 2022-12-17:
Thanks for your detailed answer! I'll close this issue for now.

closed 2022-12-17 by Kaiwen-Hong