mani-skill/ManiSkill #43: Issues when loading too many geometries

Xingyu-Lin, 2023-01-03:
I tried to import many objects into the simulation by running a load_asset function. However, after loading about 20 geometries, I got the following error:

`RuntimeError: Failed to submit geometry because geometries buffer size has been exceeded. `. 

Is there a simple way to fix this? I did call `scene.remove_articulation(obj)` after loading each object. I also past the load asset function below:

```
def _load_asset(self, urdf_path):
        loader = self.scene.create_urdf_loader()
        loader.fix_root_link = True
        loader.load_multiple_collisions_from_file = True
        density = 8e3
        obj = loader.load(str(urdf_path), config={"density": density})
        # obj.set_name("faucet")
        set_articulation_render_material(obj, color=hex2rgba("#AAAAAA"), metallic=1, roughness=0.4)
        return obj
```

fbxiang, 2023-01-04:
Since I only found this error message in Kuafu, I assume you are using KuafuRenderer and the issue is related to Kuafu, which is not a standard use case for ManiSkill.
If you need to resolve this issue immediately, I suggest posting it to Kuafu's repo and see if there is an easy fix.
If you can wait or there is no easy fix, there will be a new Kuafu release by the end of this Month, which should no longer have this problem.

Xingyu-Lin, 2023-01-04:
Thanks for the response. I will wait until the new Kuafu release then! I will find another way to get around this issue for now.

closed 2023-01-04 by Xingyu-Lin