mani-skill/ManiSkill #49: Questions about mpm state in soft body envs

shen-hhao, 2023-01-08:
Hello, I notice that you use mpm state to record particle state in soft body envs, but I don't know the exact representation of mpm state. I want to know what 'x', 'v', 'F', 'C' and 'vol' in mpm state refer to.

fbxiang, 2023-01-12:
Here is some information,
`x`: particle position
`v`: particle velocity
`F`: particle deformation gradient
`C`: spatial gradient of velocity (on particles)

shen-hhao, 2023-01-22:
[image]
Thanks for the explanation. Here is my other question.
I saved the particle position of demonstration trajectories using the trajectory replay. In this picture, blue particle is the particle of water and green particle stands for the shape of inner bottle in Pour-v0 env. This picture is about 30th frame in one demo trajectory, however the water particles seem to form into separate layers, which is different from our common sense that water drops should stay together.
I want to know if that is the flaw of simulation, or is caused by other mechanism? By the way, I noticed the water particles are spawn with large spacings.

fbxiang, 2023-01-22:
This is probably caused by the simulation, and the exact reason may require many studies. Here are some things to consider.
1. Roughly speaking, particles in MPM are not spheres, but blobs of deformed material with changing shapes. So particle distribution depends on initialization and the simulation process.
2. The current material model is not exactly modeling water, but designed to look like water.
3. The simulation is far from accurate in order to run at real time.

The large spacing is just to make sure it does not explode. If the particles are too dense (exceeding the density of water), it would explode under pressure. Placing them sparsely is generally fine.

closed 2023-01-23 by shen-hhao