mani-skill/ManiSkill #62: Error while running on A100 server. I am trying to run basic demo in A100 server, but having problems due rendering and display.

sudhirpratapyadav, 2023-02-17:
When I run `python -m mani_skill2.examples.demo_random_action` command. I get following error.

```
[2023-02-17 19:03:09.661] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing
[2023-02-17 19:03:09.661] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/examples/demo_random_action.py", line 71, in <module>
    main()
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/examples/demo_random_action.py", line 34, in main
    env: BaseEnv = gym.make(
  File "/opt/conda/lib/python3.8/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/opt/conda/lib/python3.8/site-packages/gym/envs/registration.py", line 87, in make
    env = self.entry_point(**_kwargs)
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/workspace/sudhiry/rl_ws/ManiSkill2/mani_skill2/envs/sapien_env.py", line 110, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: Cannot find a suitable rendering device
```

How to run this in a headless mode, where redering is not required?

Jiayuan-Gu, 2023-02-17:
It is usually due to the broken Vulkan driver. Can you try to run `vulkaninfo` to show whether the vulkan driver is working? (`vulkaninfo` can be installed through `sudo apt-get install vulkan-utils`)

If `vulkaninfo` fails to show the information about Vulkan, you need to create `/usr/share/vulkan/icd.d/nvidia_icd.json` and set the environment variable `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/nvidia_icd.json`.

The content of `nvidia_icd.json`:

```json
{
    "file_format_version" : "1.0.0",
    "ICD": {
        "library_path": "libGLX_nvidia.so.0",
        "api_version" : "1.2.155"
    }
}
```

fbxiang, 2023-02-17:
This should be a more comprehensive check.
https://github.com/haosulab/SAPIEN/issues/115#issuecomment-1434899965

Jiayuan-Gu, 2023-02-22:
Thank you for raising the issue. We have added more details about [troubleshooting](https://haosulab.github.io/ManiSkill2/getting_started/installation.html#vulkan) and [documentation](https://haosulab.github.io/ManiSkill2/getting_started/docker.html) about how to use ManiSkill2 in a docker.

closed 2023-02-22 by Jiayuan-Gu