mani-skill/ManiSkill #67: ImportError: libvulkan.so.1

abliao, 2023-03-05:
from .pysapien import *
ImportError: libvulkan.so.1: cannot open shared object file: No such file or directory

StoneT2000, 2024-01-16:
@gabrieletiboni @abliao take a look at https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting. Let me know if those instructions resolve the issue!

gabrieletiboni, 2024-01-16:
Thank you for your quick reply @StoneT2000 .

I'm currently trying to have ManiSkill2 work on remote cluster nodes, where libvulkan is not present. Also, I'd prefer not to use docker at the moment. As no rendering, I was hoping there would be a way to bypass the vulkan dependency altogether.

Clearly, apt installing libvulkan is not an option when sudo rights are not given.

Please let me know if you happen to know any way to do this :)

StoneT2000, 2024-01-16:
@fbxiang might know. 

But I recommend first trying to see if you can vulkan installed in any way possible. Docker is definitely a very good solution (we ourselves use docker for all our remote cluster nodes to run experiments) and if you can use it definitely do. Is there a reason why docker is not okay?

fbxiang, 2024-01-16:
To resolve this issue, you just need to install libvulkan1. I also recommend using Docker.

gabrieletiboni, 2024-01-18:
Thank you all for your replies!
Here's my hot takes:

1. Docker does introduce a bit of complexity in the process. I'm planning to add support to ManiSkill2 envs to a year-old active project, hence it would require changing the deployment / experimenting pipeline for the whole project.
2. Besides the extra work in the first point, one non-trivial point that is often overlooked is that not many people have great familiarity with Docker to begin with. You can certainly make the point that it's not the "package's fault", and more people should simply learn how to containerize their projects. Nevertheless, my personal experience in multiple research groups tells me that ~50% of PhD students / Post Doc habitually make use of Docker containers. I'm part of the less familiarized half, so I might be biased.
3. Docker isn't necessarily compatible with all compute clusters. I've worked with multiple clusters there were internally maintained at the research group level, and did not provide great compatibility with Docker containers. Bigger clusters may also not provide compatibility at all due to security reasons, but at least I heard `singularity` can help in those cases (again, extra work and knowledge = less accessibility).
4. _Any_ app is containerizable, does this mean every app _should_ be containerizable?  I don't think so, mainly for the three points above. Clearly, some packages _must_ be containerized as no workaround exists, e.g., requirements of specific system dependencies on remote clusters. However, when this is possible, I believe making the effort to make it simply pip installable without sudo rights makes it that much more accessible to the community.

For example, do ManiSkill2 envs absolutely require libvulkan when no rendering is needed? If not, a "no-rendering" option would be great at runtime, which would disable all rendering functionalities and only make use of the CPU-based underlying physics engine.

I guess this would only work for state-based observations (which is nevertheless my use case), as rendering is likely needed to compute point-clouds and depth maps anyways.

That being said, I'll probably end up going for the Docker installation regardless. Just hot takes on the "is there a reason why docker is not okay?" question :)

StoneT2000, 2024-01-19:
Thanks for the perspective on the docker issue, I understand now. At the moment our bandwidth is a bit strained between me and @fbxiang so not sure when we can try and address this. It seems like it should be possible but i'm not an expert on this side of the benchmark.

Definitely we will strive to avoid having to use sudo rights to install dependencies. I'll keep the issue open for now.

StoneT2000, 2024-01-19:
Actually, in the next sapien / maniskill, we should be able to run the environment code without vulkan, i will need to add code to automatically check if vulkan is there, and if it is not i will disable all code adding cameras. I'll update here when we have that done

kasteric, 2024-05-19:
Hi, my remote cluster does not support docker command, how can i solve this problem? can i just download the libvulkan.so.1 file on the internet and link it to the environment variables?

xuanlinli17, 2024-05-19:
please see https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#troubleshooting

StoneT2000, 2024-07-01:
Has your issue been resolved @kasteric

kasteric, 2024-07-05:
> Has your issue been resolved @kasteric ?

Hi, i didnt solve this, I later gave up headless mode, and directly ran it in my local linux enviroment with desktop.

closed 2024-11-08 by StoneT2000

litwellchi, 2024-12-10:
> @fbxiang might know.
>
> But I recommend first trying to see if you can vulkan installed in any way possible. Docker is definitely a very good solution (we ourselves use docker for all our remote [...]

Hi, I also have the same problem. Since we are using some clusters that have banned the use of docker due to some security issues, we have had to seek a solution without docker.  Is there any update on this issue?

ecielyang, 2024-12-24:
I solve this by install
`conda install conda-forge::libvulkan-loader`