mani-skill/ManiSkill #71: When robotiq 140 is used to grasp, the link will separate

ZachQianzf, 2023-03-19:
Thank you for the great benchmark. I am using the new claw robotiq 140 provided in tutorials. I found that when I take a grab action, the links of the claw will be separated. I checked the urdf as if there were no constraints between certain links. Could you provide some solutions？
[image]

Jiayuan-Gu, 2023-03-20:
Thank you for pointing it out. We currently only experimentally support xmate3+robotiq, which was used in our sim-to-real experiments in the paper. We have made simplifications about the gripper as you show above, and such simplifications are usually enough for pick-and-place tasks (quasi-static). If it is important for your project to simulate it in a more realistic way, we refer you to our CVPR tutorial: <https://www.youtube.com/watch?v=pPxft5kYcYI&t=6253s> and <https://github.com/haosulab/cvpr-tutorial-2022/blob/master/debug/robotiq.py>. We are also working on implementing it in ManiSkill2, please stay tuned.

ZachQianzf, 2023-03-21:
Thanks very much！

closed 2023-03-21 by ZachQianzf