mani-skill/ManiSkill #73: Running soft body envs on multiple GPUs

cchang99, 2023-03-29:
Hi there! I'm trying to run soft body envs on multiple GPUs, but it's only okay to run on the first card (that card has monitor plugged in). When I'm running on other cards, I did the followings:
~~~
export CUDA_VISIBLE_DEVICES=1
python -m mani_skill2.examples.demo_random_action -e Pinch-v0
~~~
But I get the following error
~~~
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
~~~

xuanlinli17, 2023-03-29:
We found that if you are running ManiSkill2 on a device where the `DISPLAY` variable is non-empty (e.g., if you are using (or vnc into) a device with a monitor connected), then `ErrorInitializationFailed` will occur regardless of rigid & soft body environments. Instead, if you ssh into the machine or if you are using a headless machine, in which case there is no `DISPLAY` variable, then you can run ManiSkill2 envs (regardless of rigid or soft body) on all GPUs.

The current temporary solution is to unset the `DISPLAY` variable by using the command `unset DISPLAY`. We'll figure out a permanent solution later.

cchang99, 2023-03-30:
OK, I run the code on the other GPU successfully! Thanks!

closed 2023-12-08 by StoneT2000