mani-skill/ManiSkill #77: Errors while create submissions

xtli12, 2023-04-03:
-  **Error  1**

When I followed the Participation Guidelines, like:
```
# Add your submission to PYTHONPATH. Ensure that "user_solution.py" can be found to import.
export PYTHONPATH=${PATH_TO_YOUR_CODES_IN_HOST}:$PYTHONPATH

# Test whether the user solution can be imported
# python -c "from user_solution import UserPolicy"

# Run evaluation. The result will be saved to ${OUTPUT_DIR}.
ENV_ID="PickCube-v0" OUTPUT_DIR="tmp"
python -m mani_skill2.evaluation.run_evaluation -e ${ENV_ID} -o ${OUTPUT_DIR}
```
The path of my `user_solution.py` is `/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/submission`
So I  add my submission to PYTHONPATH and test whether the user solution can be imported like this:
```
export PYTHONPATH=/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/submission:$PYTHONPATH
python -c "from user_solution import UserPolicy"
```
**The error comes out like this:**
```
Pointnet++ is not compiled
No module named 'maniskill2_learn.networks.modules.pn2_modules'
```
Although after I installed Pointnet++, this error still exists.

-  **Error  2**

When I tried to run a local evaluation like this:
```
ENV_ID="PickCube-v0" OUTPUT_DIR="tmp"
python -m mani_skill2.evaluation.run_evaluation -e ${ENV_ID} -o ${OUTPUT_DIR}
```
**The error comes out like this:**
```
Fail to load evaluation configuration. (<class 'TypeError'>, TypeError("'NoneType' object cannot be interpreted as an integer"))
```

-  **Error  3**

When I built my submission docker image like this:
```
docker build -f "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/Docker" -t maniskill2023-submission
```
**The error comes out like this:**
```
ERROR: "docker buildx build" requires exactly 1 argument.
See 'docker buildx build --help'.

Usage:  docker buildx build [OPTIONS] PATH | URL | -

Start a build
```
I've been trying to figure out how to fix those errors for a couple of days, but I've failed. I also can't tell if`${YOUR_CODES_AND_WEIGHTS_IN_CONTAINER}`is same as `${YOUR_CODES_AND_WEIGHTS_IN_HOST}` since I haven't found an explanation for `${YOUR_CODES_AND_WEIGHTS_IN_CONTAINER}` in the guideline.
It would be fantastic if you could assist me in resolving those errors and provide us with more detailed Participation Guidelines. Your kind assistance is much appreciated!

xuanlinli17, 2023-04-03:
Hi xtli12,

I've updated the docker files and submission examples for ManiSkill2-Learn. You can view it at https://github.com/haosulab/ManiSkill2-Learn/tree/main/submission_example . 

```
Pointnet++ is not compiled
No module named 'maniskill2_learn.networks.modules.pn2_modules'
```

This only indicates that you haven't compiled our custom implementation of pointnet++. Programs can still run successfully without PointNet++.

```
Fail to load evaluation configuration. (<class 'TypeError'>, TypeError("'NoneType' object cannot be interpreted as an integer"))
```

If you follow my updated ManiSkill2-Learn submission example, then it shouldn't occur anymore.

```
ERROR: "docker buildx build" requires exactly 1 argument.
```

The command should be 

`docker build -f {path_to_docker_file} {build_context_path} -t maniskill2023-submission`

Your command is missing the `build_context_path`. The `build_context_path` is the path where docker instructions, e.g., `COPY`, are referenced with respect to. See [here](https://docs.docker.com/engine/reference/commandline/build/ )for more details.

For example,

`docker build -f "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/Docker" . -t maniskill2023-submission`

Note the `.` in the above command.

xtli12, 2023-04-04:
Thank you for your immediate reply!
I've changed the `config.file_path` and  `model_path` in the updated `user_solution.py` as follows:
```
cfg = Config.fromfile("/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/configs/mfrl/ppo/maniskill2_pn.py")  
model_path = f'/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/model_10000000.ckpt' 
```
And moved the `user_solution.py` directly under the `ManiSkill2-Learn-main/`, but the error still exists when I run:
```
export PYTHONPATH=/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main:$PYTHONPATH
ENV_ID="PickCube-v0" OUTPUT_DIR="tmp"
python -m mani_skill2.evaluation.run_evaluation -e ${ENV_ID} -o ${OUTPUT_DIR}
```
**the error is**
```
Fail to load evaluation configuration. (<class 'TypeError'>, TypeError("'NoneType' object cannot be interpreted as an integer"))
```

xtli12, 2023-04-05:
I was running these commands outside of docker as the guideline [https://github.com/haosulab/ManiSkill2/wiki/Participation-Guidelines](url) says.
**First, I changed the `config.file_path` and `model_path` in `user_solution.py` as follows:**
```
#original
cfg = Config.fromfile("/root/ManiSkill2-Learn/configs/mfrl/ppo/maniskill2_pn.py") # Change this
model_path = f'/root/ManiSkill2-Learn/maniskill2_learn_pretrained_models_videos/{env_id}/dapg_pointcloud/model_25000000.ckpt' # Change this
```
changed to the absolute path in my machine:
```
cfg = Config.fromfile("/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/configs/mfrl/ppo/maniskill2_pn.py")  # Change this
model_path = f'/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/model_10000000.ckpt'  # Change this
```

**Then I moved the  `user_solution.py` to the directory `ManiSkill2-Learn-main/` like this:**

[image]

Then I followed the guideline [https://github.com/haosulab/ManiSkill2/wiki/Participation-Guidelines](url):
```
export PYTHONPATH=/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main:$PYTHONPATH
ENV_ID="PickCube-v0" OUTPUT_DIR="tmp"
python -m mani_skill2.evaluation.run_evaluation -e ${ENV_ID} -o ${OUTPUT_DIR}
```
**But the error still exists :**
```
Fail to load evaluation configuration. (<class 'TypeError'>, TypeError("'NoneType' object cannot be interpreted as an integer"))
```

xuanlinli17, 2023-04-05:
Oh, I believe you need to refer to the new participation guideline at https://haosulab.github.io/ManiSkill2/benchmark/submission.html . Our wiki page is obsolete. Sorry about the confusion.

xtli12, 2023-04-07:
Thank you for your reply. With your kind help, I was able to locally verify my solution !
But when I submitted the docker image to the leaderboard, my score was 0. The weird thing is that the model I employed is the official `.ckpt` like:
```
/ManiSkill2-Learn-main/maniskill2_learn_pretrained_models_videos/PickCube-v0/dapg_pointcloud_vf1/model_25000000.ckpt
```
and when I tested the docker image locally, the score was not 0, and the `average_metrics.json` file is:
```
{"elapsed_steps": 32.0, "is_obj_placed": 1.0, "is_robot_static": 1.0, "success": 1.0}
```
But the result on leaderboard is:
[image]
How could I solve it?

xuanlinli17, 2023-04-07:
Did you test the success rate locally using your submission docker?

xtli12, 2023-04-07:
Yes, I did.

StoneT2000, 2023-04-07:
@xtli12 Thanks for raising the issue. I will take a look into this issue, there may be bug in the evaluation system

StoneT2000, 2023-04-08:
@xtli12 which submission is the one that gets your success rate? I looked at your most recent submission but it seems user_solution.py is empty.
```
(base) root@abcdefg:~# cat user_solution.py 
from mani_skill2.evaluation.solution import BasePolicy

class UserPolicy(BasePolicy):
    def act(self, observations):
        return self.action_space.sample()

    @classmethod
    def get_obs_mode(cls, env_id: str) -> str:
        return "rgbd"

    @classmethod
    def get_control_mode(cls, env_id: str) -> str:
        return None # use whatever the default control mode is
```

StoneT2000, 2023-04-08:
@xtli12 we have discovered the bug. The evaluation system has been updated now. Moreover, make sure to use absolute paths in your submission

In your code I see you used some relative paths like `./....` instead of `/`. Note that all code is executed from the `/root/` directory, some of your code is in your own configured directory in `/`. I recommend using the absolute directory when possible.

Another note, make sure to not select all environment tasks to submit your image for (unless that is intended). It will slow down your evaluation, just pick the ones your code is designed for (we will also update the UI to make sure the default is not all tasks).

xtli12, 2023-04-08:
Thanks to your kind assistance, I am now able to have a score on the leaderboard!

StoneT2000, 2023-04-10:
@xtli12 actually we also fixed another relevant bug on the submission system. Some submissions weren't running on some tasks and was reporting some error about a pod. This has been fixed so your new submissions should evaluate on all tasks selected and not randomly fail.

xtli12, 2023-04-12:
@StoneT2000 
Hi, thanks for your kind assistance! However, there is still one issue that I can't figure out: When I got a `.ckpt` file after I trained locally, the success rate of it could be evaluated at nearly 0.9 (as the statistics.csv of eval_videos show):
```
length,reward,finish
16.00,27.09,1.00
30.00,55.04,1.00
27.00,46.54,1.00
21.00,39.88,1.00
18.00,25.68,1.00
32.00,70.89,1.00
200.00,54.32,0.00
200.00,142.26,0.00
19.00,33.22,1.00
21.00,38.99,1.00
25.00,37.49,1.00
25.00,41.71,1.00
28.00,47.78,1.00
22.00,36.20,1.00
36.00,62.98,1.00
22.00,42.51,1.00
26.00,47.45,1.00
20.00,38.91,1.00
16.00,27.86,1.00
15.00,15.59,1.00
24.00,38.76,1.00
34.00,56.16,1.00
23.00,39.97,1.00
16.00,24.94,1.00
28.00,55.13,1.00
18.00,28.16,1.00
29.00,26.65,1.00
24.00,41.36,1.00
22.00,33.08,1.00
17.00,26.10,1.00
32.00,52.22,1.00
23.00,47.61,1.00
22.00,28.18,1.00
20.00,30.57,1.00
14.00,26.25,1.00
36.00,54.62,1.00
21.00,39.93,1.00
25.00,45.10,1.00
27.00,41.78,1.00
36.00,65.30,1.00
24.00,40.19,1.00
35.00,53.46,1.00
25.00,46.08,1.00
38.00,64.58,1.00
27.00,48.21,1.00
31.00,49.23,1.00
35.00,57.30,1.00
33.00,55.16,1.00
30.00,57.10,1.00
29.00,51.07,1.00
22.00,39.08,1.00
17.00,32.24,1.00
29.00,53.18,1.00
21.00,31.36,1.00
31.00,51.91,1.00
30.00,54.70,1.00
17.00,30.73,1.00
24.00,42.72,1.00
16.00,27.93,1.00
25.00,46.75,1.00
21.00,36.03,1.00
20.00,38.31,1.00
26.00,50.52,1.00
34.00,71.51,1.00
18.00,34.77,1.00
21.00,32.74,1.00
[... 413 lines cut ...]
21.00,33.53,1.00
15.00,26.22,1.00
17.00,28.36,1.00
35.00,55.92,1.00
25.00,46.77,1.00
23.00,43.28,1.00
18.00,35.07,1.00
26.00,41.23,1.00
24.00,41.70,1.00
32.00,58.87,1.00
18.00,30.94,1.00
112.00,129.10,1.00
33.00,67.45,1.00
14.00,25.67,1.00
22.00,28.69,1.00
32.00,72.20,1.00
43.00,71.23,1.00
200.00,61.88,0.00
51.00,60.59,1.00
16.00,29.60,1.00
28.00,44.72,1.00
```
but when I upload it, the score is 0, maybe this is because of an overfitting problem, but it's weird that the score is totally 0, like:
[image]
only when I upload the  official pretrained `.ckpt`, the score is not 0. How can I solve this issue?

xuanlinli17, 2023-04-12:
Could you double check `user_solution.py` on e.g., are controller and observation mode correct?

xtli12, 2023-04-13:
The problem had been solved, thank you very much!

closed 2023-04-13 by xtli12