mani-skill/ManiSkill #79: Error while training under "env_name=MoveBucket-v1"

xtli12, 2023-04-10:
Hi! When I was training under "env_name=MoveBucket-v1" like:
```
python maniskill2_learn/apis/run_rl.py configs/mfrl/ppo/maniskill2_pn.py 
--work-dir /data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/Result/MoveBucket-v1/PPO 
--gpu-ids 0 --cfg-options "env_cfg.env_name=MoveBucket-v1" 
"env_cfg.obs_mode=pointcloud" "env_cfg.n_points=1200" "env_cfg.control_mode=pd_joint_delta_pos" "env_cfg.reward_mode=dense" 
"rollout_cfg.num_procs=5" "eval_cfg.num=100" "eval_cfg.save_traj=False" "eval_cfg.save_video=True" "eval_cfg.num_procs=5"

```
**there was an error:**
```
MoveBucket-v1-train - (run_rl.py:347) - INFO - 2023-04-10,03:46:16 - Set random seed to 144182442
MoveBucket-v1-train - (run_rl.py:351) - INFO - 2023-04-10,03:46:16 - Build replay buffer!
MoveBucket-v1-train - (run_rl.py:370) - INFO - 2023-04-10,03:46:16 - Build rollout!
MoveBucket-v1-train - (rollout.py:14) - INFO - 2023-04-10,03:46:16 - Rollout environments have seed from [953760095, 953760100)
Segmentation fault (core dumped)
```
But it was normal when I was training under "PickCube-v0" or "StackCube-v0", so how could I solve the error above?

xuanlinli17, 2023-04-10:
Are you running on server w/o display?

If not then you can try `DISPLAY="" python maniskill2_learn/...`

xtli12, 2023-04-10:
Thanks for your immediate reply! I am running on server with display, and I have checked the value of the `DISPLAY=""` by 
```
echo $DISPLAY
0.0
```
But when I try ` DISPLAY="0.0" python maniskill2_learn/...` or  ` DISPLAY="" python maniskill2_learn/...` the error is the same:
```
[2023-04-10 07:15:38.930] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2023-04-10 07:15:38.930] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "maniskill2_learn/apis/run_rl.py", line 487, in <module>
    main()
  File "maniskill2_learn/apis/run_rl.py", line 452, in main
    run_one_process(0, 1, args, cfg)
  File "maniskill2_learn/apis/run_rl.py", line 374, in run_one_process
    rollout = build_rollout(rollout_cfg)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/builder.py", line 15, in build_rollout
    return build_from_cfg(cfg, ROLLOUTS, default_args)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/utils/meta/registry.py", line 136, in build_from_cfg
    return obj_cls(**args)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/rollout.py", line 15, in __init__
    self.vec_env = build_vec_env(env_cfg, num_procs, seed=seed, **kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/env_utils.py", line 226, in build_vec_env
    vec_env = VectorEnv(cfgs, **vec_env_kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/vec_env.py", line 318, in __init__
    super(VectorEnv, self).__init__(env_cfgs=env_cfgs, **kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/vec_env.py", line 180, in __init__
    self.env_cfgs, self.single_env, self.num_envs = env_cfgs, build_env(env_cfgs[0]), len(env_cfgs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/env_utils.py", line 205, in build_env
    return build_from_cfg(cfg, ENVS)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/utils/meta/registry.py", line 136, in build_from_cfg
    return obj_cls(**args)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/env_utils.py", line 155, in make_gym_env
    env = gym.make(env_name, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 73, in make
    env = self.entry_point(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/base_env.py", line 55, in __init__
    super().__init__(*args, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```
but I have already solved the problem of `Vulkan` when I was training under "PickCube-v0" or "StackCube-v0", so how could I solve it?

xuanlinli17, 2023-04-10:
if you go to interactive python and run

```
import mani_skill2.envs, gym
env=gym.make('MoveBucket-v0', obs_mode='rgbd')
obs=env.reset()
```

Does it core dump?

xtli12, 2023-04-10:
When I go to interactive python and run
```
import mani_skill2.envs, gym
env=gym.make('MoveBucket-v0', obs_mode='rgbd')
obs=env.reset()
```
it shows like:
```
>>> import mani_skill2.envs, gym
>>> env=gym.make('MoveBucket-v0', obs_mode='rgbd')
Traceback (most recent call last):
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 150, in spec
    return self.env_specs[id]
KeyError: 'MoveBucket-v0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 105, in make
    spec = self.spec(path)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 161, in spec
    raise error.DeprecatedEnv(
gym.error.DeprecatedEnv: Env MoveBucket-v0 not found (valid versions include ['MoveBucket-v1'])
>>> obs=env.reset()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'env' is not defined
```
does it mean I haven't  downloaded the `MoveBucket-v0`? But I have already downloaded the `MoveBucket-v0` before by
```
python -m mani_skill2.utils.download_asset partnet_mobility_bucket
```

xuanlinli17, 2023-04-10:
oh sorry, it should be MoveBucket-v1, not v0

xtli12, 2023-04-10:
it shows initialization failed:
```
>>> import mani_skill2.envs, gym
>>> env=gym.make('MoveBucket-v1', obs_mode='rgbd')
Segmentation fault (core dumped)
```

xuanlinli17, 2023-04-10:
Try 

display=“” python

xtli12, 2023-04-10:
It shows like:
```
>>> DISPLAY=""
>>> 
```

xuanlinli17, 2023-04-10:
I mean 

DISPLAY=“” python

then create movebucket ebb in the interactive puthon

xtli12, 2023-04-10:
It shows like:
```
>>> DISPLAY=""
>>> import mani_skill2.envs, gym
>>> env=gym.make('MoveBucket-v1', obs_mode='rgbd')
[2023-04-10 08:38:11.157] [svulkan2] [error] GLFW error: X11: Failed to open display “”
[2023-04-10 08:38:11.157] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 73, in make
    env = self.entry_point(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/base_env.py", line 55, in __init__
    super().__init__(*args, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```

xuanlinli17, 2023-04-10:
https://github.com/haosulab/ManiSkill2/issues/73#issuecomment-1489473119

linking this as a reference

xuanlinli17, 2023-04-10:
Oh, you need to first exit the interactive python, then 

DISPLAY=“” python

to reenter the interactive Python with display env variable set to empty,

then create movebucket env inside the interactive python

xtli12, 2023-04-10:
It shows like:
```
(sapien) lxt21@ubuntu:~/SAPIEN-master/ManiSkill2-Learn-main$ DISPLAY="" python
Python 3.8.0 (default, Nov  6 2019, 21:49:08) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mani_skill2.envs, gym
>>> env=gym.make('MoveBucket-v1', obs_mode='rgbd')
[2023-04-10 08:43:04.682] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2023-04-10 08:43:04.682] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 73, in make
    env = self.entry_point(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/base_env.py", line 55, in __init__
    super().__init__(*args, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```

xuanlinli17, 2023-04-10:
Did you ensure that the 3 nvidia json files exist according to https://haosulab.github.io/ManiSkill2/getting_started/installation.html#vulkan

xtli12, 2023-04-10:
Yes, the 3 nvidia json files already existed.
[image]
[image]
[image]

xuanlinli17, 2023-04-10:
Do they have the right content?

xtli12, 2023-04-10:
The content differences between them are in api_version:
[image]
[image]
After I changed both of them to`"api_version" : "1.2.155"`, the error still occurred:
```
(sapien) lxt21@ubuntu:~/SAPIEN-master/ManiSkill2-Learn-main$ DISPLAY="" python
Python 3.8.0 (default, Nov  6 2019, 21:49:08) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mani_skill2.envs, gym
>>> env=gym.make('MoveBucket-v1', obs_mode='rgbd')
[2023-04-10 09:11:02.929] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2023-04-10 09:11:02.929] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 184, in make
    return registry.make(id, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 106, in make
    env = spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/envs/registration.py", line 73, in make
    env = self.entry_point(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/base_env.py", line 55, in __init__
    super().__init__(*args, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorInitializationFailed
```

xtli12, 2023-04-10:
Oh, I have found that the error `Segmentation fault (core dumped)` occurs when there are too many tasks on one GPU. When I stop some tasks, it is solved, but when I rerun:
```
python maniskill2_learn/apis/run_rl.py configs/mfrl/ppo/maniskill2_pn.py 
--work-dir /data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/Result/MoveBucket-v1/PPO 
--gpu-ids 0 --cfg-options "env_cfg.env_name=MoveBucket-v1" 
"env_cfg.obs_mode=pointcloud" "env_cfg.n_points=1200" "env_cfg.control_mode=pd_joint_delta_pos" "env_cfg.reward_mode=dense" 
"rollout_cfg.num_procs=5" "eval_cfg.num=100" "eval_cfg.save_traj=False" "eval_cfg.save_video=True" "eval_cfg.num_procs=5"
```
New error occurred:
```
AssertionError: pd_joint_delta_pos not in supported modes: ['base_pd_joint_vel_arm_pd_joint_vel', 
'base_pd_joint_vel_arm_pd_joint_delta_pos', 'base_pd_joint_vel_arm_pd_joint_pos', 
'base_pd_joint_vel_arm_pd_joint_target_delta_pos', 'base_pd_joint_vel_arm_pd_ee_delta_pos', 
'base_pd_joint_vel_arm_pd_ee_delta_pose', 'base_pd_joint_vel_arm_pd_ee_target_delta_pos', 
'base_pd_joint_vel_arm_pd_ee_target_delta_pose', 'base_pd_joint_vel_arm_pd_joint_pos_vel', 
'base_pd_joint_vel_arm_pd_joint_delta_pos_vel']
```
So I changed "env_cfg.control_mode=pd_joint_delta_pos" to "env_cfg.control_mode=base_pd_joint_vel_arm_pd_joint_vel" , but another error occurred:
```
Traceback (most recent call last):
  File "maniskill2_learn/apis/run_rl.py", line 487, in <module>
    main()
  File "maniskill2_learn/apis/run_rl.py", line 452, in main
    run_one_process(0, 1, args, cfg)
  File "maniskill2_learn/apis/run_rl.py", line 374, in run_one_process
    rollout = build_rollout(rollout_cfg)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/builder.py", line 15, in build_rollout
    return build_from_cfg(cfg, ROLLOUTS, default_args)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/utils/meta/registry.py", line 136, in build_from_cfg
    return obj_cls(**args)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/rollout.py", line 15, in __init__
    self.vec_env = build_vec_env(env_cfg, num_procs, seed=seed, **kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/env_utils.py", line 226, in build_vec_env
    vec_env = VectorEnv(cfgs, **vec_env_kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/vec_env.py", line 318, in __init__
    super(VectorEnv, self).__init__(env_cfgs=env_cfgs, **kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/vec_env.py", line 188, in __init__
    self.buffers = create_buffer_for_env(self.single_env, self.num_envs, self.SHARED_NP_BUFFER)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/vec_env.py", line 44, in create_buffer_for_env
    obs = env.reset()
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/wrappers.py", line 97, in reset
    obs = self.env.reset(*args, **kwargs)
  File "/data/home-gxu/lxt21/.conda/envs/sapien/lib/python3.8/site-packages/gym/wrappers/time_limit.py", line 27, in reset
    return self.env.reset(**kwargs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/wrappers.py", line 222, in reset
    return self.observation(obs)
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main/maniskill2_learn/env/wrappers.py", line 377, in observation
    base_pose = observation["agent"]["base_pose"]
KeyError: 'base_pose'
```
How could I solve the error above?

xuanlinli17, 2023-04-11:
I've just fixed ManiSkill2-Learn. Please pull the latest code.

xtli12, 2023-04-17:
Hi, I could run my code under "env_name=MoveBucket-v1" with the `control_mode=base_pd_joint_vel_arm_pd_joint_delta_pos`, However, when I uploaded the Docker image to the leaderboard, I encountered an issue:

[image]
I'm sure my code can run locally and evluated locally, and I've double check `user_solution.py`. What steps should I take to resolve this issue?(The problem only occurs when I run my code with Articulation tasks)

xuanlinli17, 2023-04-17:
Did you use the latest maniskill2-learn? 

Please make sure the maniskill2-learn codebase you are importing from is the latest.

xtli12, 2023-04-19:
Thank you for your generous help, the issue of `KeyError: 'tcp_pose'` has been solved! However, I am facing another problem: I only can use the GPU[0] for training, when I changed `--gpu-ids 0` to `--gpu-ids 1` it came with:

```
RuntimeError: Cannot find cuda device suitable for rendering cuda:1
```
even when I change to `--gpu-ids 0 --sim-gpu-ids 1`  or add `CUDA_VISIBLE_DEVICES=1` before my traing code, the error still persists. What steps should I take to resolve this issue?

xtli12, 2023-04-20:
when I try:
```
DISPLAY="" CUDA_VISIBLE_DEVICES=2 python debug.py 
```
it came out:
[image]

and when I put `DISPLAY="" CUDA_VISIBLE_DEVICES=2` before my training script, the error `[error] GLFW error: X11: Failed to open display` still persists, but it starts training on GPU[2]. Dose the error `[error] GLFW error: X11: Failed to open display` matter?

xuanlinli17, 2023-04-20:
It doesn't matter.

xtli12, 2023-04-20:
Thank you much! By the way, there is some uncertainty around the timing of verifying the list of entries.  Specifically, I'm not sure whether I should submit the list of entries after the prize results are published, or confirm the list of entries during the challenge period.

StoneT2000, 2023-04-20:
@xtli12 if you are referring to the final entry selection on your team page, you must do that before the competition ends.

xtli12, 2023-04-20:
Got it, thank you very much.

xuanlinli17, 2023-04-21:
It says "no such file or directory", so I guess

(1) modify the pycharm debug config file, since the root directory might be wrong
(2) ensure the correct ide and project path
(3) in maniskill2-learn, pass in `rollout_cfg.num_procs=1` and `eval_cfg.num_procs=1`, to avoid debugging parallel envs.

xtli12, 2023-04-22:
Hi, does the leaderboard only display 100 past submissions? I uploaded No.101, but it didn't appear on the leaderboard. If it's limited to displaying only 100 past submissions, how can I delete a historical submission?

Also, I'm uncertain about whether the final result is evaluated using a single docker image that can fit all tasks in one track, or if it's a summary of previous submissions based on many individual docker images, as shown on the current leaderboard?

StoneT2000, 2023-04-22:
This seems to be a bug with the pagination, Ill have that fixed over the weekend @xtli12 thanks for finding this.

And yes you are right, we ask you to put all your best models over two submissions (we will only rigorously evaluate two submissions for second stage). 

I will also add a button to clear the final submission selection so you don't need to scroll too far.

xtli12, 2023-04-22:
Thanks for your prompt reply! I think it would be better to increase the final submission limit so that it can include all tasks (for example, 14 submissions in the rigid-body track and 6 submissions in the soft-body track), because according to the guidelines, it seems like we cannot train on multiple environments, and I am unsure about how to combine all the best models into one Docker image and evaluate them successfully.

xtli12, 2023-05-05:
Hi, has the submission issue been resolved? @StoneT2000 

> Hi, does the leaderboard only display 100 past submissions? I uploaded No.101, but it didn't appear on the leaderboard. If it's limited to displaying only 100 past submissions, how can I delete a historical submission?

StoneT2000, 2023-05-05:
Not just yet, sorry some other issues have raised that took precedence. Will aim to fix this weekend

StoneT2000, 2023-05-08:
@xtli12 new features have been added. Pagination is now fixed so it'll fetch the right data when you go to the next page, and you can sort most columns across all of the database (e.g. sort your submissions by success rate and it will refetch data accordingly). Moreover, you can clear all final submission selections now in one click to avoid having to go through all the pages (we disable sorting by which submissions are final submissions atm since it is not working with the new pagination). Let me know if you have any issues!

Also, we are still discussing about how many docker images to allow for submission so stay tuned.

xtli12, 2023-05-09:
Thank you for your kind assistance! I'll keep an eye on the issue regarding how many docker images are allowed for submission. However, there is a small problem with the leaderboard page. When I uploaded the No.101 submission, the No.1 submission disappeared, and the page only shows the latest 100 submissions (No.2-No.101). Therefore, it may be not possible to select the best submission from history when making the final submission. It would be better to add a button to delete historical submissions.

StoneT2000, 2023-05-10:
you should be able to sort by date, does it not show the latest one?

xtli12, 2023-05-11:
yes,  it can show oldest one now, thank you very much!

StoneT2000, 2023-05-17:
> Thanks for your prompt reply! I think it would be better to increase the final submission limit so that it can include all tasks (for example, 14 submissions in the rigid-body track and 6 [...]

we have bumped up the max number of final submissions per track to 14 now, so you can have one image per task instead.

xtli12, 2023-05-19:
Ok, thank you very much!

closed 2023-05-19 by xtli12