mani-skill/ManiSkill #80: Adding new robots ?

jrabary, 2023-04-19:
Hi, 
Thanks for making maniskill open. We are really interesting in using it to develop algorithms for robotic manipulation. But we'd like to deploy the policy on some low-cost robot like https://www.trossenrobotics.com/robotic-arms/ros-research-arms.aspx. Gazebo simulation exists for them. We are wondering how many effort will be needed to add them inside maniskill and if you can provide support for that.

xuanlinli17, 2023-04-20:
Hi jrabary,

You can take a look at https://github.com/haosulab/ManiSkill2/blob/main/examples/tutorials/customize_environments.ipynb . There is a tutorial section for importing custom robots into maniskill2.

StoneT2000, 2023-12-08:
https://github.com/haosulab/ManiSkill2/pull/175 will include a new, easier-to-read/easier-to-work-with robot API and some documentation around how robots are added. To be merged soon-ish

closed 2024-02-21 by StoneT2000