mani-skill/ManiSkill #82: Google Colab Your session crashed for an unknown reason

megatran, 2023-04-24:
Hello! I'm running into an issue when trying to run the provided `1_quickstart.ipynb`  on Google Colab. My runtime has GPU. Is anyone encountering this as well?

[image]

Here's the log

|Timestamp|Level|Message|
|---|---|---|
|Apr 24, 2023, 9:58:50 AM|WARNING|WARNING:root:kernel 015636f9-1898-49cf-9191-a9efcd9d7ae8 restarted|
|Apr 24, 2023, 9:58:50 AM|INFO|KernelRestarter: restarting kernel \(1/5\), keep random ports|
|Apr 24, 2023, 9:58:48 AM|WARNING|\[2023-04-24 13:58:48\.303\] \[svulkan2\] \[error\] Some required Vulkan extension is not present\. You may not use the renderer to render, however, CPU resources will be still available\.|
|Apr 24, 2023, 9:58:48 AM|WARNING|\[2023-04-24 13:58:48\.298\] \[svulkan2\] \[warning\] Continue without GLFW\.|
|Apr 24, 2023, 9:58:48 AM|WARNING|\[2023-04-24 13:58:48\.298\] \[svulkan2\] \[error\] GLFW error: X11: The DISPLAY environment variable is missing|
|Apr 24, 2023, 9:48:47 AM|INFO|Kernel started: 015636f9-1898-49cf-9191-a9efcd9d7ae8, name: python3|
|Apr 24, 2023, 9:48:34 AM|INFO|Use Control-C to stop this server and shut down all kernels \(twice to skip confirmation\)\.|
|Apr 24, 2023, 9:48:34 AM|INFO|http://172\.28\.0\.12:9000/|
|Apr 24, 2023, 9:48:34 AM|INFO|Jupyter Notebook 6\.4\.8 is running at:|
|Apr 24, 2023, 9:48:34 AM|INFO|Serving notebooks from local directory: /|
|Apr 24, 2023, 9:48:33 AM|INFO|Use Control-C to stop this server and shut down all kernels \(twice to skip confirmation\)\.|
|Apr 24, 2023, 9:48:33 AM|INFO|http://172\.28\.0\.2:9000/|
|Apr 24, 2023, 9:48:33 AM|INFO|Jupyter Notebook 6\.3\.0 is running at:|
|Apr 24, 2023, 9:48:33 AM|INFO|Serving notebooks from local directory: /|
|Apr 24, 2023, 9:48:33 AM|WARNING|    ModuleNotFoundError: No module named 'jupyter\_server'|
|Apr 24, 2023, 9:48:33 AM|WARNING|        from jupyter\_server\.base\.handlers import JupyterHandler|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "/usr/local/lib/python3\.9/dist-packages/panel/io/jupyter\_server\_extension\.py", line 53, in \<module\>|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "\<frozen importlib\.\_bootstrap\>", line 228, in \_call\_with\_frames\_removed|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "\<frozen importlib\.\_bootstrap\_external\>", line 850, in exec\_module|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "\<frozen importlib\.\_bootstrap\>", line 680, in \_load\_unlocked|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "\<frozen importlib\.\_bootstrap\>", line 986, in \_find\_and\_load\_unlocked|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "\<frozen importlib\.\_bootstrap\>", line 1007, in \_find\_and\_load|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "\<frozen importlib\.\_bootstrap\>", line 1030, in \_gcd\_import|
|Apr 24, 2023, 9:48:33 AM|WARNING|        return \_bootstrap\.\_gcd\_import\(name\[level:\], package, level\)|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "/usr/lib/python3\.9/importlib/\_\_init\_\_\.py", line 127, in import\_module|
|Apr 24, 2023, 9:48:33 AM|WARNING|        mod = importlib\.import\_module\(modulename\)|
|Apr 24, 2023, 9:48:33 AM|WARNING|      File "/usr/local/lib/python3\.9/dist-packages/notebook/notebookapp\.py", line 2033, in init\_server\_extensions|
|Apr 24, 2023, 9:48:33 AM|WARNING|    Traceback \(most recent call last\):|
|Apr 24, 2023, 9:48:33 AM|WARNING|Error loading server extension panel\.io\.jupyter\_server\_extension|
|Apr 24, 2023, 9:48:29 AM|INFO|google\.colab serverextension initialized\.|
|Apr 24, 2023, 9:48:29 AM|INFO|google\.colab serverextension initialized\.|
|Apr 24, 2023, 9:48:29 AM|INFO|Authentication of /metrics is OFF, since other authentication is disabled\.|
|Apr 24, 2023, 9:48:29 AM|INFO|Writing notebook server cookie secret to /root/\.local/share/jupyter/runtime/notebook\_cookie\_secret|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/root/\.jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/root/\.local/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/usr/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/usr/local/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/usr/local/etc/jupyter/jupyter\_notebook\_config\.d/panel-client-jupyter\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|INFO|Authentication of /metrics is OFF, since other authentication is disabled\.|
|Apr 24, 2023, 9:48:29 AM|INFO|Writing notebook server cookie secret to /root/\.local/share/jupyter/runtime/notebook\_cookie\_secret|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/root/\.jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/root/\.local/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/usr/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/usr/local/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/usr/local/etc/jupyter/jupyter\_notebook\_config\.d/panel-client-jupyter\.json|
|Apr 24, 2023, 9:48:29 AM|WARNING|    	/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.825 NotebookApp\] Loaded config file: /root/\.jupyter/jupyter\_notebook\_config\.py|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.818 NotebookApp\] Looking for jupyter\_notebook\_config in /root/\.jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.816 NotebookApp\] Looking for jupyter\_notebook\_config in /root/\.local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.816 NotebookApp\] Looking for jupyter\_notebook\_config in /usr/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.815 NotebookApp\] Loaded config file: /usr/local/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.812 NotebookApp\] Looking for jupyter\_notebook\_config in /usr/local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.812 NotebookApp\] Loaded config file: /etc/jupyter/jupyter\_notebook\_config\.py|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.808 NotebookApp\] Looking for jupyter\_notebook\_config in /etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.806 NotebookApp\] Looking for jupyter\_config in /root/\.jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.795 NotebookApp\] Looking for jupyter\_config in /root/\.local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.795 NotebookApp\] Looking for jupyter\_config in /usr/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.777 NotebookApp\] Looking for jupyter\_config in /usr/local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.776 NotebookApp\] Looking for jupyter\_config in /etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.769 NotebookApp\] Searching \['/root/\.jupyter', '/root/\.local/etc/jupyter', '/usr/etc/jupyter', '/usr/local/etc/jupyter', '/etc/jupyter'\] for config files|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.769 NotebookApp\] Loaded config file: /root/\.jupyter/jupyter\_notebook\_config\.py|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.765 NotebookApp\] Looking for jupyter\_notebook\_config in /root/\.jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.764 NotebookApp\] Looking for jupyter\_notebook\_config in /root/\.local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.764 NotebookApp\] Looking for jupyter\_notebook\_config in /usr/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.764 NotebookApp\] Loaded config file: /usr/local/etc/jupyter/jupyter\_notebook\_config\.json|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.762 NotebookApp\] Looking for jupyter\_notebook\_config in /usr/local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.761 NotebookApp\] Loaded config file: /etc/jupyter/jupyter\_notebook\_config\.py|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.759 NotebookApp\] Looking for jupyter\_notebook\_config in /etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.758 NotebookApp\] Looking for jupyter\_config in /root/\.jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.758 NotebookApp\] Looking for jupyter\_config in /root/\.local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.757 NotebookApp\] Looking for jupyter\_config in /usr/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.742 NotebookApp\] Looking for jupyter\_config in /usr/local/etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.738 NotebookApp\] Looking for jupyter\_config in /etc/jupyter|
|Apr 24, 2023, 9:48:28 AM|WARNING|\[D 13:48:28\.738 NotebookApp\] Searching \['/root/\.jupyter', '/root/\.local/etc/jupyter', '/usr/etc/jupyter', '/usr/local/etc/jupyter', '/etc/jupyter'\] for config files|

StoneT2000, 2023-04-24:
Taking a look

megatran, 2023-04-25:
Thanks @StoneT2000. I wonder if there are any recent package upgrades that might have broken this. I know Google Colab sometimes likes to upgrade Python version so I tried to downgrade it to 3.9 and 3.8, still crashed.

skylar93, 2023-04-25:
Having the same issue. :(

xuanlinli17, 2023-04-25:
For now could you try `!pip install https://storage1.ucsd.edu/wheels/sapien-dev/sapien-2.0.0.dev20230405-cp39-cp39-manylinux2014_x86_64.whl` in the install commands in the colab?

megatran, 2023-04-26:
> 

I can confirm that this is the fix. Thanks so much @xuanlinli17

ankile, 2023-04-26:
These were the imports and setup lines that seemingly fixed it for me:

```python
!pip install jedi==0.16
!pip install setuptools==65.6.3
!apt-get install -y --no-install-recommends libvulkan-dev
!pip install mani_skill2 stable_baselines3
!pip install --upgrade --no-cache-dir gdown
!pip install pyvirtualdisplay

!apt-get install -y xvfb python-opengl

!pip install https://storage1.ucsd.edu/wheels/sapien-dev/sapien-2.0.0.dev20230405-cp39-cp39-manylinux2014_x86_64.whl

from pyvirtualdisplay import Display

display = Display(visible=0, size=(1024, 768))
display.start()
```

Hopefully, this can save you from an hour or two of agony!

StoneT2000, 2023-04-26:
Update: It seems colab now is missing some files shown in https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting

A temporary solution we will consider now is having colab notebooks download the missing files and copy them over from github.

megatran, 2023-04-27:
> Update: It seems colab now is missing some files shown in https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting
> 
> A temporary solution we will consider now is having colab notebooks download the missing files and copy them over from github.

Thanks @StoneT2000 ...this issue is happening again today. I tried to follow your suggestion by download and placing the files in colab

```
!sudo apt-get install libglvnd-dev
!sudo mkdir -p /usr/share/vulkan/icd.d
!sudo wget -q https://raw.githubusercontent.com/haosulab/ManiSkill2/main/docker/nvidia_icd.json
!sudo mv nvidia_icd.json /usr/share/vulkan/icd.d
!cat /usr/share/vulkan/icd.d/nvidia_icd.json

!sudo mkdir -p /etc/vulkan/implicit_layer.d
!sudo wget -q https://raw.githubusercontent.com/haosulab/ManiSkill2/main/docker/nvidia_layers.json 
!sudo mv nvidia_layers.json /etc/vulkan/implicit_layer.d/
!cat /etc/vulkan/implicit_layer.d/nvidia_layers.json

!sudo mkdir -p /usr/share/glvnd/egl_vendor.d  # Create dir if it doesn't exist 

!echo '{ "file_format_version" : "1.0.0", "ICD" : { "library_path" : "libEGL_nvidia.so.0" } }' > /usr/share/glvnd/egl_vendor.d/10_nvidia.json   

!cat /usr/share/glvnd/egl_vendor.d/10_nvidia.json 

!sudo apt-get install vulkan-utils

!mkdir -p /tmp/runtime  # Create directory 
!export XDG_RUNTIME_DIR=/tmp/runtime  # Set environment variable
!vulkaninfo
```

Now I'm getting

```
error: XDG_RUNTIME_DIR not set in the environment.
/build/vulkan-tools-KEbD_A/vulkan-tools-1.2.131.1+dfsg1/vulkaninfo/vulkaninfo.h:926: failed with ERROR_UNKNOWN
```

Any suggestions?

megatran, 2023-04-27:
Also, getting

```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
[<ipython-input-67-30d5f10bdeda>](https://localhost:8080/#) in <cell line: 19>()
     17 
     18 # create an environment with our configs and then reset to a clean state
---> 19 env = gym.make(env_id, obs_mode=obs_mode, reward_mode=reward_mode, control_mode=control_mode)
     20 obs = env.reset()
     21 print("Action Space:", env.action_space)

11 frames
[/usr/local/lib/python3.10/dist-packages/mani_skill2/sensors/camera.py](https://localhost:8080/#) in __init__(self, camera_cfg, scene, renderer_type, articulation)
    141         # Add camera
    142         if self.actor is None:
--> 143             self.camera = scene.add_camera(
    144                 camera_cfg.uid,
    145                 camera_cfg.width,

RuntimeError: copmileToSpirv: failed to compile shader
```

StoneT2000, 2023-04-27:
I will be out today, I should be able to push a proper fix by tomorrow hopefully. One thing you could try first is just make sure you follow all steps in the trouble shooting document and make sure to reset the kernel.

megatran, 2023-04-27:
> I will be out today, I should be able to push a proper fix by tomorrow hopefully. One thing you could try first is just make sure you follow all steps in the trouble shooting document and make sure to reset the kernel.

Thanks so much for the quick reply. I did follow the troubleshooting section to copy the missing files here

https://github.com/haosulab/ManiSkill2/issues/82#issuecomment-1526576361

xuanlinli17, 2023-04-28:
Now it should be fixed.

closed 2023-04-28 by StoneT2000