mani-skill/ManiSkill #84: Regarding the environment dimensions

skylar93, 2023-05-01:
Hi,
When I get the demos from the dataset using this code, the observation and action dimensions are: 
```
dataset = ManiSkill2Dataset(f"demos/rigid_body/{env_id}/trajectory.state.pd_ee_delta_pose.h5")
dataloader = DataLoader(dataset, batch_size=256, num_workers=0, pin_memory=True, drop_last=True, shuffle=True)
obs, action = dataset[0]
print("Observation:", obs.shape)
print("Action:", action.shape)

Observation: torch.Size([42])
Action: torch.Size([7])
```

However, when I reset the environment, the observation and action dimension looks like 75 and 12. 

```
obs_mode = "state"
env = gym.make("OpenCabinetDrawer-v1", obs_mode=obs_mode)
obs = env.reset() 
obs dimension (75, )
```

I was wondering if the other 33 dimensions contain some other information. I read through the documents, but couldn't really understand the discrepancy. Could you please let me know about the mismatch?

skylar93, 2023-05-02:
Hi, 

Thanks for your reply. I'm trying to match the environment up right now.

```
env_id = "OpenCabinetDrawer-v1"

!python -m mani_skill2.utils.download_asset partnet_mobility_cabinet

# download the raw dataset
!python -m mani_skill2.utils.download_demo {env_id} -o demos 

# Directly download the converted demonstrations dataset files
!gdown "https://drive.google.com/drive/folders/1oNLgMVkByoHMv6mio-KxTkd7DrE6amKZ?usp=share_link" --folder -O "demos/rigid_body/{env_id}"

Retrieving folder list
Processing file 1QkkKDKfoJgu2Mv02687vFMEsGiJRkgx3 trajectory.h5
Processing file 1LfZPUN88izYPl7sjd6KK7FZSQnE8zleo trajectory.json
Processing file 1zrO6YDCOym8XfJbsg43H6VXkwqNpF9_Z trajectory.rgbd.pd_ee_delta_pose.h5
Processing file 1nInnP1uw9dyUuKdCwl_mtzAnnlbEoyso trajectory.rgbd.pd_ee_delta_pose.json
Processing file 1RSvvATA1UtPfA6lMJvdBXS4UY2Nn10-v trajectory.state.pd_ee_delta_pose.h5
Processing file 1kUxQfGp-gycODCxRnV6uulpnyxmIYHdJ trajectory.state.pd_ee_delta_pose.json
```

As you suggested, I used the trajectory.h5 file because everything else had the pd_ee_delta_pose on it. 

```
# convert the dataset to include state observations.
!python -m mani_skill2.trajectory.replay_trajectory --traj-path \
    demos/rigid_body/{env_id}/trajectory.h5 --save-traj \
    --obs-mode state --target-control-mode pd_joint_vel --num-procs 10
```

```
dataset = ManiSkill2Dataset(f"demos/rigid_body/{env_id}/trajectory.state.pd_joint_vel.h5")
dataloader = DataLoader(dataset, batch_size=256, num_workers=0, pin_memory=True, drop_last=True, shuffle=True)
obs, action = dataset[0]
Observation: torch.Size([42])
Action: torch.Size([8])
```

I get 42 and 8. For the env.reset() however, I get:

```
obs_mode = "state"
control_mode = "base_pd_joint_vel_arm_pd_joint_vel"
env = gym.make("OpenCabinetDrawer-v1", obs_mode=obs_mode, control_mode=control_mode)
obs = env.reset()
obs.shape (75, )
```

When I look into the document, 

```
    if args.env_id in MS1_ENV_IDS:
        if args.control_mode is not None and not args.control_mode.startswith("base"):
            args.control_mode = "base_pd_joint_vel_arm_" + args.control_mode

```

this suggests putting the pd_joint_vel is sufficient to make the control_mode to  "base_pd_joint_vel_arm_pd_joint_vel". Is there anything wrong in the order of the codes or how I should format them?

xuanlinli17, 2023-05-02:
`https://drive.google.com/drive/folders/1oNLgMVkByoHMv6mio-KxTkd7DrE6amKZ`

This google drive folder corresponds to `LiftCube-v0`, not `OpenCabinetDrawer-v1`.

xuanlinli17, 2023-05-02:
You can see the list of files here: `https://drive.google.com/drive/u/1/folders/1pd9Njg2sOR1VSSmp-c1mT7zCgJEnF8r7`

closed 2023-09-24 by xuanlinli17