mani-skill/ManiSkill #85: Google Colab failed "No module named 'mani_skill2'"

megatran, 2023-05-02:
Hello!

It looks like Google Colab probably updates again...now the second tutorial on RL fails

https://colab.research.google.com/github/haosulab/ManiSkill2/blob/main/examples/tutorials/2_reinforcement_learning.ipynb

This import fails `import mani_skill2.envs`

the pip installation log is below

```
Looking in indexes: https://pypi.org/simple, https://us-python.pkg.dev/colab-wheels/public/simple/
Collecting setuptools==65.5.0
  Downloading setuptools-65.5.0-py3-none-any.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 15.2 MB/s eta 0:00:00
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 67.7.2
    Uninstalling setuptools-67.7.2:
      Successfully uninstalled setuptools-67.7.2
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
ipython 7.34.0 requires jedi>=0.16, which is not installed.
cvxpy 1.3.1 requires setuptools>65.5.1, but you have setuptools 65.5.0 which is incompatible.
Successfully installed setuptools-65.5.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libvulkan-dev is already the newest version (1.2.131.2-1).
libvulkan-dev set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 24 not upgraded.
Looking in indexes: https://pypi.org/simple, https://us-python.pkg.dev/colab-wheels/public/simple/
Collecting mani_skill2
  Downloading mani_skill2-0.4.2-py3-none-any.whl (12.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 12.4/12.4 MB 60.4 MB/s eta 0:00:00
Collecting stable_baselines3
  Downloading stable_baselines3-1.8.0-py3-none-any.whl (174 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 174.5/174.5 kB 20.3 MB/s eta 0:00:00
Requirement already satisfied: numpy<1.24 in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (1.22.4)
Requirement already satisfied: scipy in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (1.10.1)
Collecting sapien==2.2.1
  Downloading sapien-2.2.1-cp39-cp39-manylinux2014_x86_64.whl (39.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 39.1/39.1 MB 14.7 MB/s eta 0:00:00
Requirement already satisfied: tqdm in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (4.65.0)
Requirement already satisfied: tabulate in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (0.8.10)
Collecting trimesh
  Downloading trimesh-3.21.5-py3-none-any.whl (680 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 680.8/680.8 kB 61.5 MB/s eta 0:00:00
Collecting gym<=0.21.0,>=0.18.3
  Downloading gym-0.21.0.tar.gz (1.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.5/1.5 MB 76.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: h5py in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (3.8.0)
Requirement already satisfied: imageio[ffmpeg] in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (2.25.1)
Collecting GitPython
  Downloading GitPython-3.1.31-py3-none-any.whl (184 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 184.3/184.3 kB 24.6 MB/s eta 0:00:00
Collecting transforms3d
  Downloading transforms3d-0.4.1.tar.gz (1.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.4/1.4 MB 85.0 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: opencv-python in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (4.7.0.72)
Requirement already satisfied: pyyaml in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (6.0)
Collecting rtree
  Downloading Rtree-1.0.1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.0/1.0 MB 73.4 MB/s eta 0:00:00
Requirement already satisfied: gdown>=4.6.0 in /usr/local/lib/python3.9/dist-packages (from mani_skill2) (4.6.6)
Requirement already satisfied: requests>=2.22 in /usr/local/lib/python3.9/dist-packages (from sapien==2.2.1->mani_skill2) (2.27.1)
Requirement already satisfied: cloudpickle in /usr/local/lib/python3.9/dist-packages (from stable_baselines3) (2.2.1)
Requirement already satisfied: torch>=1.11 in /usr/local/lib/python3.9/dist-packages (from stable_baselines3) (2.0.0+cu118)
Requirement already satisfied: pandas in /usr/local/lib/python3.9/dist-packages (from stable_baselines3) (1.5.3)
[... 84 lines cut ...]
    Found existing installation: gdown 4.6.6
    Uninstalling gdown-4.6.6:
      Successfully uninstalled gdown-4.6.6
Successfully installed gdown-4.7.1
Looking in indexes: https://pypi.org/simple, https://us-python.pkg.dev/colab-wheels/public/simple/
Collecting sapien==2.0.0.dev20230405
  Downloading https://storage1.ucsd.edu/wheels/sapien-dev/sapien-2.0.0.dev20230405-cp39-cp39-manylinux2014_x86_64.whl (39.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 39.1/39.1 MB 13.6 MB/s eta 0:00:00
Requirement already satisfied: transforms3d>=0.3 in /usr/local/lib/python3.9/dist-packages (from sapien==2.0.0.dev20230405) (0.4.1)
Requirement already satisfied: opencv-python>=4.0 in /usr/local/lib/python3.9/dist-packages (from sapien==2.0.0.dev20230405) (4.7.0.72)
Requirement already satisfied: requests>=2.22 in /usr/local/lib/python3.9/dist-packages (from sapien==2.0.0.dev20230405) (2.27.1)
Requirement already satisfied: numpy>=1.17 in /usr/local/lib/python3.9/dist-packages (from sapien==2.0.0.dev20230405) (1.22.4)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/local/lib/python3.9/dist-packages (from requests>=2.22->sapien==2.0.0.dev20230405) (1.26.15)
Requirement already satisfied: charset-normalizer~=2.0.0 in /usr/local/lib/python3.9/dist-packages (from requests>=2.22->sapien==2.0.0.dev20230405) (2.0.12)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.9/dist-packages (from requests>=2.22->sapien==2.0.0.dev20230405) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.9/dist-packages (from requests>=2.22->sapien==2.0.0.dev20230405) (2022.12.7)
Installing collected packages: sapien
  Attempting uninstall: sapien
    Found existing installation: sapien 2.2.1
    Uninstalling sapien-2.2.1:
      Successfully uninstalled sapien-2.2.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
mani-skill2 0.4.2 requires sapien==2.2.1, but you have sapien 2.0.0.dev20230405 which is incompatible.
Successfully installed sapien-2.0.0.dev20230405
```

StoneT2000, 2023-05-02:
Damn for some reason colab keeps changing. For example, my colab is suddenly now in python 3.10. I'm currently trying to investigate this but I think this may just be a call for #76 to be completed which includes a gymnasium upgrade. Gym has some bugs that just break with newer versions of setuptools and probably the importlibmetadata package, which are both auto installed and upgraded by Colab.

StoneT2000, 2023-05-02:
https://stackoverflow.com/questions/76129688/why-is-pip-install-gym-failing-with-python-setup-py-egg-info-did-not-run-succ we are not alone :/

megatran, 2023-05-02:
thank you for the quick reply 😭😭😭 would you recommend setting a local environment with python 3.8 meanwhile?

StoneT2000, 2023-05-02:
Yeah if you have your own GPU I recommend a local environment using Conda.

StoneT2000, 2023-05-02:
Here's a new fix. apparently git cloning the v0.21.0 version of gym and then doing a local install works

```
!mkdir -p /usr/share/vulkan/icd.d
!wget -q https://raw.githubusercontent.com/haosulab/ManiSkill2/main/docker/nvidia_icd.json
!wget -q https://raw.githubusercontent.com/haosulab/ManiSkill2/main/docker/10_nvidia.json
!mv nvidia_icd.json /usr/share/vulkan/icd.d
!mv 10_nvidia.json /usr/share/glvnd/egl_vendor.d/10_nvidia.json
# dependencies
!pip install setuptools==65.5.0
!apt-get install -y --no-install-recommends libvulkan-dev
!git clone --depth 1 --branch v0.21.0 https://github.com/openai/gym.git
!pip install -e gym
!pip install mani_skill2 stable_baselines3
!pip install --upgrade --no-cache-dir gdown
```

this is now the new setup code for colab specifically, you can try this @megatran 

I'll also update this on all tutorials.

megatran, 2023-05-02:
you guys are the best. Thanks so much @StoneT2000 !!! It works!

So while waiting for the colab fix, I attempted to run maniskills2 on a gpu server and ran into a different problem. I created a separate issue here https://github.com/haosulab/ManiSkill2/issues/86

closed 2023-05-02 by megatran