mani-skill/ManiSkill #86: mani_skill2 - ERROR - 'NoneType' object has no attribute 'vertices'

megatran, 2023-05-02:
Hello!

Since there's some current issue with Google Colab (https://github.com/haosulab/ManiSkill2/issues/85), I decided to switch to a gpu server (ssh from my local mac, I'm using XQuartz and X11 forwarding)

My test jupyter notebook seems to render the env fine
```
```

megatran, 2023-05-02:
updated with more details

StoneT2000, 2023-05-02:
Does this still fail when you try on colab after the new fix? Or this is just your GPU server

megatran, 2023-05-02:
Colab runs fine (I'm training the Visual RL block at the moment). 

The GPU setup uses `srun` to request for resources.

I'm ssh to it from my Mac (using -X forwarding with XQuartz for graphics). After a GPU resource is allocated, I can ssh into that and I can confirm that X11 forwarding still works by using `xclock` in the server and the clock actually shows up on my Mac screen.

I'm using conda env with Python 3.8

My ~/.bashrc in the server has these configuration for vulkan

```
export VK_ICD_FILENAMES=/etc/vulkan/icd.d/nvidia_icd.json
export VK_LAYER_PATH=/etc/vulkan/implicit_layer.d/nvidia_layers.json
```

Since the regular env can run and render the scene, I think the graphical forwarding is working.

[image]

However, this only happens when I try to use `VecEnv`. There's this error/warning that says `Only 1 renderer is allowed per process. All previously created renderer resources are now invalid`. This makes me wonder whether the GPU-optimized vectorized environments are trying to create multiple Vulkan renderes within a single process? From the error, I interpret it as Vulkan only allowing "1 render per process" but this `VecEnv` is somehow attempting to create multiple renderers/resources per process.

megatran, 2023-05-02:
So I'm running into a similar issue when I try on a local linux laptop with a GPU

Here's my local Python 3.8 env setup

```
Package                  Version    Editable project location
------------------------ ---------- -------------------------------------------
absl-py                  1.4.0
antlr4-python3-runtime   4.9.3
anyio                    3.6.2
argon2-cffi              21.3.0
argon2-cffi-bindings     21.2.0
arrow                    1.2.3
asttokens                2.2.1
attrs                    23.1.0
backcall                 0.2.0
beautifulsoup4           4.12.2
bleach                   6.0.0
cachetools               5.3.0
certifi                  2022.12.7
cffi                     1.15.1
charset-normalizer       3.1.0
cloudpickle              2.2.1
cmake                    3.26.3
comm                     0.1.3
contourpy                1.0.7
cycler                   0.11.0
debugpy                  1.6.7
decorator                5.1.1
defusedxml               0.7.1
executing                1.2.0
fastjsonschema           2.16.3
filelock                 3.12.0
fonttools                4.39.3
fqdn                     1.5.1
gdown                    4.7.1
gitdb                    4.0.10
GitPython                3.1.31
google-auth              2.17.3
google-auth-oauthlib     1.0.0
grpcio                   1.54.0
gym                      0.21.0
h5py                     3.8.0
hydra-core               1.3.2
idna                     3.4
imageio                  2.28.1
imageio-ffmpeg           0.4.8
importlib-metadata       4.13.0
importlib-resources      5.12.0
ipykernel                6.22.0
ipython                  8.12.1
ipython-genutils         0.2.0
isoduration              20.11.0
jedi                     0.18.2
Jinja2                   3.1.2
jsonpointer              2.3
jsonschema               4.17.3
jupyter_client           8.2.0
jupyter_core             5.3.0
jupyter-events           0.6.3
jupyter_server           2.5.0
jupyter_server_terminals 0.4.4
jupyterlab-pygments      0.2.2
kiwisolver               1.4.4
lit                      16.0.2
mani-skill2              0.4.2      /home/nt/Documents/RobotLearning/ManiSkill2
Markdown                 3.4.3
MarkupSafe               2.1.2
matplotlib               3.7.1
matplotlib-inline        0.1.6
[... 218 lines cut ...]
    206 # Infer texture names
    207 texture_names = set()

File ~/miniconda3/envs/robotlearning38/lib/python3.8/multiprocessing/connection.py:250, in _ConnectionBase.recv(self)
    248 self._check_closed()
    249 self._check_readable()
--> 250 buf = self._recv_bytes()
    251 return _ForkingPickler.loads(buf.getbuffer())

File ~/miniconda3/envs/robotlearning38/lib/python3.8/multiprocessing/connection.py:414, in Connection._recv_bytes(self, maxsize)
    413 def _recv_bytes(self, maxsize=None):
--> 414     buf = self._recv(4)
    415     size, = struct.unpack("!i", buf.getvalue())
    416     if size == -1:

File ~/miniconda3/envs/robotlearning38/lib/python3.8/multiprocessing/connection.py:379, in Connection._recv(self, size, read)
    377 remaining = size
    378 while remaining > 0:
--> 379     chunk = read(handle, remaining)
    380     n = len(chunk)
    381     if n == 0:

ConnectionResetError: [Errno 104] Connection reset by peer

```

megatran, 2023-05-02:
after further debugging, I wonder if the opencabinet env is the culprit,

When I default to the example, Colab, GPU server, and local machine seem to run fine!

```
num_envs = 2 # you can increases this and decrease the n_steps parameter if you have more cores to speed up training
env_id = "LiftCube-v0"
obs_mode = "state"
control_mode = "pd_ee_delta_pose"
reward_mode = "dense"
```

StoneT2000, 2023-05-02:
Could you try using a sparse reward setting for the failing envs?

StoneT2000, 2023-05-02:
Closing in favor of https://github.com/haosulab/ManiSkill2/issues/88

StoneT2000, 2023-05-02:
Current solution: https://github.com/haosulab/ManiSkill2/issues/88#issuecomment-1532194498

closed 2023-05-02 by StoneT2000