mani-skill/ManiSkill #90: Error when runing sb3_ppo_liftcube_rgbd

jiaming-robot-learning, 2023-05-10:
Hi,

I run the example rl code 
```
python examples/tutorials/reinforcement-learning/sb3_ppo_liftcube_rgbd.py
```
but get the following error: 
======
Traceback (most recent call last):
  File "/home/user/Projects/manipulation/mani_env/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/user/Projects/manipulation/mani_env/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/Projects/manipulation/stable-baselines3/stable_baselines3/common/vec_env/subproc_vec_env.py", line 30, in _worker
    env = _patch_env(env_fn_wrapper.var())
                     ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Projects/manipulation/ManiSkill2/examples/tutorials/reinforcement-learning/sb3_ppo_liftcube_rgbd.py", line 293, in make_env
    env = RecordEpisode(
          ^^^^^^^^^^^^^^
  File "/home/user/Projects/manipulation/ManiSkill2/mani_skill2/utils/wrappers/record.py", line 136, in __init__
    self.render_mode = render_mode
    ^^^^^^^^^^^^^^^^
AttributeError: property 'render_mode' of 'RecordEpisode' object has no setter
Traceback (most recent call last):
  File "/home/user/Projects/manipulation/ManiSkill2/examples/tutorials/reinforcement-learning/sb3_ppo_liftcube_rgbd.py", line 405, in <module>
    main()
  File "/home/user/Projects/manipulation/ManiSkill2/examples/tutorials/reinforcement-learning/sb3_ppo_liftcube_rgbd.py", line 312, in main
    eval_env = SubprocVecEnv([env_fn for _ in range(1)])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Projects/manipulation/stable-baselines3/stable_baselines3/common/vec_env/subproc_vec_env.py", line 121, in __init__
    observation_space, action_space = self.remotes[0].recv()
                                      ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Projects/manipulation/mani_env/lib/python3.11/multiprocessing/connection.py", line 249, in recv
    buf = self._recv_bytes()
          ^^^^^^^^^^^^^^^^^^
  File "/home/user/Projects/manipulation/mani_env/lib/python3.11/multiprocessing/connection.py", line 413, in _recv_bytes
    buf = self._recv(4)
          ^^^^^^^^^^^^^
  File "/home/user/Projects/manipulation/mani_env/lib/python3.11/multiprocessing/connection.py", line 378, in _recv
    chunk = read(handle, remaining)

==================

(My environment is newly set up)

StoneT2000, 2023-05-10:
Could you try using python 3.9? And re run it

jiaming-robot-learning, 2023-05-19:
that worked. Thanks