mani-skill/ManiSkill #91: Do you have exhaustive results on your benchmark tasks?

HeegerGao, 2023-05-10:
Hi, ManiSkill2 is such a wonderful work and I am very interested in it! I'd like to know if there are any exhaustive results on all of your tasks in this benchmark, with every possible IL/RL algorithm and input format. In your paper, I can only find the results of:
1. BC
2. PPO+DAPG

But obviously, there are many other mainstream algos to be tested. In fact, the most thing I want to know are results with **pure RL algorithms with point-cloud-based input** on all of your tasks. It would be great if you could provide these results. Thanks!

StoneT2000, 2023-05-10:
Thanks for bringing this up @HeegerGao. Testing many other algorithms is a massive effort that we want to undertake. That being said we currently can't test all the new algorithms. Is there any particular ones you had in mind? We do plan to add more baselines in the future (especially CleanRL single file style ones)

xuanlinli17, 2023-05-10:
PPO should be able to solve PickCube, LiftCube, and StackCube. For most other tasks, success rates are low or near zero (even though state-based policy can solve e.g., PegInsertionSide, PlugCharger when training sufficiently long (>10M)). We'll update the rewards in 0.5.0, which are better suited for RL training purposes.

HeegerGao, 2023-05-11:
> PPO should be able to solve PickCube, LiftCube, and StackCube. For most other tasks, success rates are low or near zero (even though state-based policy can solve e.g., PegInsertionSide, PlugCharger when training sufficiently long (>10M), but not for vision-based policies). We'll update the rewards in 0.5.0, which are better suited for RL training purposes.

@StoneT2000 @xuanlinli17 Thank you very much for your quick reply! I will wait for your 0.5.0 version.

You said "PPO should be able to solve PickCube, LiftCube, and StackCube." I want to know if I can reproduce these results with the default configs of [Maniskill2-Learn](https://github.com/haosulab/ManiSkill2-Learn), with the `env_cfg.obs_frame=ee` trick, the `env_cfg.n_goal_points=50` trick, and using the `pd_ee_delta_pose` controller. Are there any other tricks?

xuanlinli17, 2023-05-11:
LiftCube is easy.
For PickCube and StackCube, use a large number of steps (25M+). PickCube will start to succeed after like 10M+. StackCube might need even more steps.

Hilbert-Johnson, 2023-06-12:
> PPO should be able to solve PickCube, LiftCube, and StackCube. For most other tasks, success rates are low or near zero (even though state-based policy can solve e.g., PegInsertionSide, PlugCharger [...]

Hi! Thanks for the effort the author makes! Here I want to ask that, whether you have try state-based policy using PPO/SAC on all tasks (only rigid task is also ok)? Can vanilla PPO algorithm with fully-observed state and the dense reard function you provided solve all the tasks?

xuanlinli17, 2023-06-12:
Yes. state-based SAC is capable of solving PickCube, LiftCube, StackCube easily; PegInsertionSide needs >10M to solve; PlugCharger needs >15M; for Turnfaucet and PickSingleYCB/EGAD, state-based policy can solve some objects but not all (since it doesn't have vision and thus doesn't understand object geometry), similar for all ManiSkill1 envs; for AssemblingKits, state-based policy does not achieve >0% performance as far as I know.

Hilbert-Johnson, 2023-06-12:
> Yes.
>
> State-based SAC is capable of solving PickCube, LiftCube, StackCube easily; PegInsertionSide needs >10M to solve; PlugCharger needs >15M; For Turnfaucet and PickSingleYCB/EGAD, state-based [...]

Thanks for your timely reply! How about the 4 Mobile Manipulation tasks? And can you give some insights on the SAC/PPO parameters you used to try these tasks?

xuanlinli17, 2023-06-12:
For MS1 envs (mobile manipulation), state-based SAC should be able to solve most of single object envs.

Hilbert-Johnson, 2023-06-12:
> For MS1 envs (mobile manipulation), state-based SAC should be able to solve most of single object envs. You can use the ManiSkill2-Learn parameters.

Thanks! You mention **single object**, does this mean I should only train on one fixed Cabinet mesh, instead of using different random mesh? I notice that every time after `env.reset()`, the object mesh will have an intra-class change. If so, how can I fix the object?

xuanlinli17, 2023-06-12:
For MS1 envs, if you type in `print(env.model_db)`, it will print out all asset ids:

```
>>> env=gym.make('OpenCabinetDoor-v1')
>>> print(env.model_db)
{'1000': {'num_target_links': 2, 'partnet_mobility_id': 1000, 'scale': 0.6129399556808358}, '1001': {'num_target_links': 1, 'partnet_mobility_id': 1001, 'scale': 0.512205865781575}, '1002': {'num_target_links': 1, 'partnet_mobility_id': 1002, 'scale': 0.4520109061191429}, ..., '1081': {'num_target_links': 1, 'partnet_mobility_id': 1081, 'scale': 0.43974655521394934}}
```

Then, you can choose the specific subset of asset ids by creating the environment as follows:
```
>>> env=gym.make('OpenCabinetDoor-v1', model_ids=['1000', '1001'])
```