mani-skill/ManiSkill #92: RuntimeError: Cannot find a suitable rendering device

ZiLianC, 2023-05-11:
After I install the package, I run the following command:
python -m mani_skill2.examples.demo_random_action

opts: []
env_kwargs: {}
Traceback (most recent call last):
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/examples/demo_random_action.py", line 71, in <module>
    main()
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/examples/demo_random_action.py", line 34, in main
    env: BaseEnv = gym.make(
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/gym/envs/registration.py", line 87, in make
    env = self.entry_point(**_kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/ps/anaconda3/envs/maniskill/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 102, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: Cannot find a suitable rendering device

xuanlinli17, 2023-05-11:
Do you have GPU on your device and try `nvidia-smi`?

ZiLianC, 2023-05-11:
Yes. My configuration is as follows
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.103.01   Driver Version: 470.103.01   CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
|  0%   45C    P8    38W / 350W |   2381MiB / 24265MiB |      2%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

xuanlinli17, 2023-05-11:
Could you troubleshoot vulkan following https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting

And ensure that the 3 nvidia json files exist?

ZiLianC, 2023-05-11:
The files `/usr/share/vulkan/icd.d/nvidia_icd.json` and `/usr/share/glvnd/egl_vendor.d/10_nvidia.json` don't exist. I create the files as the tutorial. After I run the command 'sudo apt-get install libglvnd-dev', the file `/usr/share/glvnd/egl_vendor.d/10_nvidia.json` is still not exist. So I create it manually. But the above problem still exist.

xuanlinli17, 2023-05-11:
```
sudo apt-get install vulkan-utils
vulkaninfo
```

Does this produce any info?

ZiLianC, 2023-05-11:
I run `vulkaninfo`, and the error is as follow:

ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object file: No such file or directory
ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object file: No such file or directory
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.

xuanlinli17, 2023-05-11:
Could you reinstall cuda driver following https://developer.nvidia.com/cuda-11.3.0-download-archive?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=runfile_local

use local runfile

xuanlinli17, 2023-05-11:
Could you reinstall your nvidia driver, using newer versions like `nvidia-driver-530`?

xuanlinli17, 2023-05-24:
Has the issue been resolved?

closed 2023-09-24 by xuanlinli17