mani-skill/ManiSkill #95: Error from replaying trajectory of "PlugCharger-v0"

FredaZero, 2023-05-20:
Hi,

I am training a model under "env=PlugCharger-v0", I wanted to convert the control mode from pd_joint_pos to pd_ee_delta_pos by running the replay_trajectory file, but it was unsuccessful, and it's the same story for replaying "PegInsertionSide-v0". Does it mean it's unchangeable? If I want to use end-effector control, do I have to manually compute the results from joint control data?

xuanlinli17, 2023-05-20:
Could you post your script here? If you follow the scripts in ManiSkill2-Learn it shouldn't happen.

FredaZero, 2023-05-21:
> Could you post your script here? If you follow the scripts in ManiSkill2-Learn it shouldn't happen.

Thanks for replying.

I converted other task demonstrations to my desired control mode successfully with the following code:
`python -m mani_skill2.trajectory.replay_trajectory --traj-path ./demos/rigid_body/PegInsertionSide-v0/trajectory.h5 -o rgbd -c pd_ee_delta_pos --save-traj`

But when I am trying to convert PegInsertinSide-v0, it turns out:
[image]

xuanlinli17, 2023-05-21:
Ok, I believe it's because you used `pd_ee_delta_pos` (3d ee xyz + 1d gripper), which can fail on some tasks where gripper rotation is required (e.g., PegInsertionSide, PlugCharger). However, if you use `pd_ee_delta_pose` (6d ee + 1d gripper), then it will be successful.

FredaZero, 2023-05-21:
> Ok, I believe it's because you used `pd_ee_delta_pos` (3d ee xyz + 1d gripper), which can fail on some tasks where gripper rotation is required (e.g., PegInsertionSide, PlugCharger). However, if you use `pd_ee_delta_pose` (3d ee xyz + 3d ee rotation + 1d gripper), then it will be successful.

Thank you! It works now!

closed 2023-05-21 by FredaZero