mani-skill/ManiSkill #101: Errors about training with Soft-body

xtli12, 2023-06-01:
Training issue:
When I run the script:
```
CUDA_VISIBLE_DEVICES=0 python maniskill2_learn/apis/run_rl.py configs/brl/bc/pointnet_soft_body.py --work-dir ../ManiSkill2-Learn-main/Result/Fill/PPO_pointnet --gpu-ids 0 --cfg-options "env_cfg.env_name=Fill-v0" "env_cfg.obs_mode=pointcloud" "env_cfg.control_mode=pd_joint_delta_pos" "eval_cfg.num=100" "eval_cfg.num_procs=4" "eval_cfg.save_traj=False" "eval_cfg.save_video=True" "replay_cfg.buffer_filenames=../data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-new/soft_body/Fill-v0/trajectory.none.pd_joint_delta_pos_pointcloud.h5"  --resume-from /data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-main-old/maniskill2_learn_pretrained_models_videos/Fill_pcd/model_final.ckpt
```
the error is:
```
Fill-v0-train - (run_rl.py:288) - INFO - 2023-06-01,05:06:07 - Train over GPU [0]!
Traceback (most recent call last):
  File "maniskill2_learn/apis/run_rl.py", line 487, in <module>
    main()
  File "maniskill2_learn/apis/run_rl.py", line 452, in main
    run_one_process(0, 1, args, cfg)
  File "maniskill2_learn/apis/run_rl.py", line 432, in run_one_process
    main_rl(rollout, evaluator, replay, args, cfg, expert_replay=expert_replay, recent_traj_replay=recent_traj_replay)
  File "maniskill2_learn/apis/run_rl.py", line 293, in main_rl
    train_rl(
  File "/data/home-gxu/lxt21/SAPIEN-master/ManiSkill2-Learn-new/maniskill2_learn/apis/train_rl.py", line 185, in train_rl
    all_shape = replay.memory.shape
AttributeError: 'NoneType' object has no attribute 'shape'
```

xtli12, 2023-06-01:
Oh, I had mistaked the buffer_filenames path, now checking the submission...

xtli12, 2023-06-01:
I may have encountered the same issue as before under `env_name=OpenCabinetDoor-v1`. I am able to have a score outside of the Docker image locally, but when running within the Docker image (using the latest version of ManiSkill2), the score is 0. The submission ID for this issue is `647866f9e09b0d48eff26846`. Is it possible that the cause of this issue is:

> Ok checked the submission. The numerical result shown (0.488, 0.04) are indeed correct. Your policy is close to opening many of the test assets but just falls a little short. I tried with a few other seeds and the values range from 0.04 to 0.08, so there's some amount of randomness here.
> 
> The percentage completion issue can be ignored, it's likely some race condition with how the progress bars are updated and potentially network issues.
> 
> It also seems that there may have some non deterministic parts that weren't handled. The server now has all evaluations seeded properly.

xtli12, 2023-06-02:
I may have encountered the same issue as before under `env_name=OpenCabinetDoor-v1`. I am able to have a score outside of the Docker image locally, but when running within the Docker image (using the latest version of ManiSkill2), the score is 0. The submission ID for this issue is `647866f9e09b0d48eff26846`. Is it possible that the cause of this issue is:

> After private discussion with the original poster, the conclusion is as follows so far
> 
> The scores on the leaderboard are indeed correct and the leaderboard is evaluating submissions correctly. The problem lies in the user_solution.py file which leverages ManiSkill2-Learn. Currently OpenCabinetDoor and the other ManiSkill1 environments do not return agent base_pose values to the observation. To address that the ManiSkill2-Learn framework automatically adds it by using the actual environment's robot information.
> 
> However, in submissions, the user_solution.py file creates a dummy environment to process observations which includes adding the base_pose. This dummy environment is not the actual environment so there is a mismatch in results as running the ManiSkill2-Learn evaluation will produce successes but running the actual evaluation where there is no access to the actual environment will cause errors as the base_pose values are wrong (and this is also why the robot looks like its only going one direction, the observation is not really changing)
> 
> We are currently discussing fixes for this.

StoneT2000, 2023-06-03:
I'll verify today/tomorrow, but it's not immediately clear at the moment

xtli12, 2023-06-04:
Hi, has the stage 2 of evaluation begun?

xtli12, 2023-06-04:
Is June 3rd, 2023 the deadline?
[image]

xuanlinli17, 2023-06-04:
It’s June 12

xtli12, 2023-06-04:
OK，but my submission has stuck for about 13 hours, what's the problem?
[image]

xuanlinli17, 2023-06-14:
Communicated and resolved on 6/5.

closed 2023-06-14 by xuanlinli17