mani-skill/ManiSkill #102: Maniskill2 and ManiSkill2-learn on headless GPU nodes in HPC environment?

cousins, 2023-06-16:
Hi,

I'm trying to help someone get ManiSkill2 and ManiSkill2-Learn running on our HPC system using GPU nodes. I assume that the general use of the GPUs is for processing but it also appears that there is some expectation of doing some rendering, using a X-Windows display. First, is there a way to decouple these? So that someone can do the GPU processing on a headless GPU node that has no rendering capabilities (we have A100 cards) and then use a different system to do the rendering?

I saw mention of the A100 card in your documentation so I was hopeful. The only JSON file in the /etc/vulkan directory is:

`/etc/vulkan/implicit_layer.d/nvidia_layers.json`

I saw some mention of running without a DISPLAY variable set so I tried the example:

```
Singularity> DISPLAY="" python -m mani_skill2.examples.demo_random_action -e PickCube-v0

opts: []
env_kwargs: {}
[2023-06-16 12:54:03.457] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2023-06-16 12:54:03.457] [svulkan2] [warning] Continue without GLFW.
[2023-06-16 12:54:03.459] [svulkan2] [error] Some required Vulkan extension is not present. You may not use the renderer to render, however, CPU resources will be still available.
Segmentation fault (core dumped)
```

I have also looked through the Issues here on Github but it doesn't seem that the relevant issues have been resolved. 

Can you please help me to figure out how to get this going?

Thanks,

Steve

xuanlinli17, 2023-06-16:
Hi Steve,

Could you ensure that the 3 jsons file exist according to https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting ?

cousins, 2023-06-16:
Only the one JSON file that I mentioned was installed. I'm using the
haosulab/mani-skill2 Docker container. Is there a container that is more up
to date?

xuanlinli17, 2023-06-16:
Probably you can build a docker here?

https://github.com/haosulab/ManiSkill2/blob/main/docker/Dockerfile

We'll also update the official docker.

xuanlinli17, 2023-06-18:
The official docker (`docker pull haosulab/mani-skill2:latest`) contains all 3 json files on my end.

xuanlinli17, 2023-06-20:
Does

```
singularity pull maniskill2-latest.sif docker://haosulab/mani-skill2:latest
```

work for you?

xuanlinli17, 2023-06-20:
The json files are here: `https://github.com/haosulab/ManiSkill2/tree/main/docker`

If you enter `ManiSkill2/docker` and then `docker build . -t {}` then it should add the json files.

JankowskiChristopher, 2024-04-06:
@cousins were you able to solve your issue?
I've just got the same error as you. I do not have sudo on the cluster and GPUs also are headless. I tried running the official Docker image through singularity, but got the same errors as you.
I asked our admin to install Vulkan, but he could not add the json config files as they would conflict with other cluster functionalities and without them and when not using the Docker image I get such error:
```
RuntimeError: vk::Instance::enumeratePhysicalDevices: ErrorInitializationFailed
```

StoneT2000, 2024-04-06:
Those json config files are absolutely required for rendering @JankowskiChristopher, there is currently no way around it. I am a little confused why those json config files can't be added however, what kind of conflicts are there?

closed 2025-05-20 by StoneT2000