mani-skill/ManiSkill #103: Is there any plan to support cloth-like object manipulation environments in the future?

Jeffery-Zhou, 2023-06-25:
Hey, there. 
Glad to see a wonderful simulation tool for versatile manipulation tasks. I saw soft-body is already supported in the environments, and I am trying to customise an environment for other cloth-like objects manipulation task, so I was wondering if current Maniskill2 has such features and would you like to give any hints about that? Thanks.

xuanlinli17, 2023-06-25:
Currently ManiSkill2 doesn't support cloth manipulation, but we plan to support it in the future, probably in a few months @fbxiang

Jeffery-Zhou, 2023-06-26:
Nice.

youliangtan, 2025-01-31:
Hi any updates on the support for cloth-like object manipulation?

StoneT2000, 2025-02-05:
Hi @youliangtan we are still working on cloth simulation but as most of our recent focus shifted to GPU sim it was deprioritized to an extent. I unfortunately don't have a clean timeline at this time