mani-skill/ManiSkill #104: Error from soft-body environment

seann999, 2023-06-26:
I think I was able to build warp with no problems yet I get this error

```
$ python3 -m mani_skill2.trajectory.replay_trajectory --traj-path demos/soft_body/Excavate-v0/trajectory.h5 --vis
0step [00:00, ?step/s]Warp initialized:
   Version: 0.3.1
   CUDA device: NVIDIA GeForce RTX 3090
   Kernel cache: /home/sean/.cache/warp/0.3.1
2023-06-26 11:00:35,142 - mani_skill2 - INFO - load sdfs from file
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
...
File "/home/sean/.local/lib/python3.8/site-packages/warp_maniskill/warp/context.py", line 1009, in zeros
    raise RuntimeError("Memory allocation failed on device: {} for {} bytes".format(device, num_bytes))
RuntimeError: Memory allocation failed on device: cuda for 120 bytes
0step [00:03, ?step/s]
```

I'm running it on cuda 11.8 with a 3090.

xuanlinli17, 2023-06-26:
try `CUDA_VISIBLE_DEVICES=0 python3 -m {cmd}`? Soft body envs can only be run on a single device.

seann999, 2023-06-26:
Thanks for the quick response.
I'm getting the same error. I think I only have one device; here's my nvidia-smi:

```
$ nvidia-smi
Mon Jun 26 15:01:21 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.105.01   Driver Version: 515.105.01   CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
|  0%   44C    P8    35W / 350W |   1216MiB / 24576MiB |     36%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1142      G   /usr/lib/xorg/Xorg                 66MiB |
|    0   N/A  N/A      1903      G   /usr/lib/xorg/Xorg                356MiB |
|    0   N/A  N/A      2035      G   /usr/bin/gnome-shell               38MiB |
|    0   N/A  N/A      2502    C+G   ...481965522237065556,131072      560MiB |
|    0   N/A  N/A      2686      G   ...RendererForSitePerProcess      125MiB |
|    0   N/A  N/A    870260      G   gnome-control-center               14MiB |
|    0   N/A  N/A   2969696      G   ...2gtk-4.0/WebKitWebProcess       15MiB |
|    0   N/A  N/A   2970172      G   /usr/bin/vlc                       18MiB |
+-----------------------------------------------------------------------------+
```

xuanlinli17, 2023-06-27:
Hard to figure out the reason...

We'd suggest to restart computer, reinstall the driver, and reinstall ManiSkill2

quantumiracle, 2023-10-30:
same error here

quantumiracle, 2023-10-30:
> same error here

Just found that my case is caused by a trivial mis-operaiton. 

In compiling warp with ``export PYTHONPATH=/path/to/ManiSkill/warp_maniskill:$PYTHONPATH``, the path is not replaced, so the compiled warp is saved somewhere else that later program cannot find.

Hope this may help