mani-skill/ManiSkill #109: could you tell me where the PD controller  code ?

lijinming2018, 2023-07-13:
I hope to deploy the agent trained by maniskill2 to a real machine .is there any documentation or code to learn ?

xuanlinli17, 2023-07-13:
It's here: https://github.com/haosulab/ManiSkill2/tree/main/mani_skill2/agents/controllers

xuanlinli17, 2023-07-13:
Documentation: https://haosulab.github.io/ManiSkill2/concepts/controllers.html

lijinming2018, 2023-07-14:
Thank you for your prompt reply

lijinming2018, 2023-07-14:
>  PD controller: control loop based on $F(t) = K_p (\bar{q}(t) - q(t)) + K_d (\bar{\dot{q}}(t) - \dot{q}(t))$. $K_p$ (stiffness) and $K_d$ (damping) are hyperparameters. $F(t)$ is the force of the motor. 

I want to know the number of the   $K_p$  and $K_d$  .  If I want to carry on a real robot experiment, How can I implement the PD controller method mentioned here?

xuanlinli17, 2023-07-14:
Stiffness and damping parameters are in the agent config files:

e.g., for stationary panda:

https://github.com/haosulab/ManiSkill2/blob/fc08823bf96791946591a508f336d724fa7cad26/mani_skill2/agents/configs/panda/defaults.py#L33

lijinming2018, 2023-07-28:
What are the units of the corresponding dimensions for each action

xuanlinli17, 2023-07-28:
See 

Actions [-1,1] are remapped to the lower and upper limits. Joint positions are in rads. ee pos is in meters. ee rotation is in rads.

lijinming2018, 2023-07-29:
Thankyou

closed 2023-07-29 by xuanlinli17

kaijieshi7, 2024-05-03:
@xuanlinli17 .Hi, Can I ask you a question? If the real robot can reach the target position in given time, do i need to care about the damping and stiffness parameters in simulation? For example, the robot hand's finger can reach the target position in 0.2 seconds in real life.

xuanlinli17, 2024-05-03:
In this case if the control is blocking, then you can just use the default parameters I think. As long as the joints reach the target position and matches the real behavior, it doesn't matter too much.