mani-skill/ManiSkill #110: When I run code on multiple GPUs, I get this error."

lijinming2018, 2023-07-13:
I run this code 
‘’‘
>python maniskill2_learn/apis/run_rl.py configs/brl/bc/rgbd.py \> --work-dir log/ceshi --gpu-ids 0 1  \> --cfg-options >"env_cfg.env_name=PickCube-v0" "env_cfg.obs_mode=rgbd" "env_cfg.n_points=1200" \
> "env_cfg.control_mode=pd_ee_delta_pose" \
> "replay_cfg.buffer_filenames=../ManiSkill2/demos/pickcube_trajectory_new1000.h5" \
> "eval_cfg.num=100" "eval_cfg.save_traj=False" "eval_cfg.save_video=False" \
> "train_cfg.n_eval=50000" "train_cfg.total_steps=50000" "train_cfg.n_checkpoint=50000" "train_cfg.n_updates=500"

’‘’
get： 

> ...
> '''
> PickCube-v0-train-0 - (run_rl.py:346) - INFO - 2023-07-13,23:48:32 - Config:
> agent_cfg = dict(
>     type='BC',
>     batch_size=256,
>     actor_cfg=dict(
>         type='ContinuousActor',
>         head_cfg=dict(type='TanhHead', noise_std=1e-05),
>         nn_cfg=dict(
>             type='Visuomotor',
>             visual_nn_cfg=dict(
>                 type='IMPALA',
>                 in_channel='image_channels',
>                 image_size='image_size',
>                 out_feature_size=384),
>             mlp_cfg=dict(
>                 type='LinearMLP',
>                 norm_cfg=None,
>                 mlp_spec=['384 + agent_shape', 256, 128, 'action_shape'],
>                 bias=True,
>                 inactivated_output=True)),
>         optim_cfg=dict(type='Adam', lr=0.0003)))
> env_cfg = dict(
>     type='gym',
>     env_name='PickCube-v0',
>     unwrapped=False,
>     obs_mode='rgbd',
>     n_points=1200,
>     control_mode='pd_ee_delta_pose',
>     device='cuda:0')
> replay_cfg = dict(
>     type='ReplayMemory',
>     capacity=8888,
>     num_samples=-1,
>     keys=['obs', 'actions', 'dones', 'episode_dones'],
>     buffer_filenames='../ManiSkill2/demos/pickcube_trajectory_new1000.h5')
> train_cfg = dict(
>     on_policy=False,
>     total_steps=50000,
>     warm_steps=0,
>     n_steps=0,
>     n_updates=500,
>     n_eval=50000,
>     n_checkpoint=50000)
> eval_cfg = dict(
>     type='Evaluation',
>     num=100,
>     num_procs=1,
>     use_hidden_state=False,
>     save_traj=False,
>     save_video=False,
>     use_log=False)
> work_dir = None
> resume_from = None
> expert_replay_cfg = None
> recent_traj_replay_cfg = None
> rollout_cfg = None
> 
[... 145 lines cut ...]
> Traceback (most recent call last):
>   File "/opt/conda/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 69, in _wrap
>     fn(i, *args)
>   File "/data/private/ljm/ManiSkill2-Learn/maniskill2_learn/apis/run_rl.py", line 432, in run_one_process
>     main_rl(rollout, evaluator, replay, args, cfg, expert_replay=expert_replay, recent_traj_replay=recent_traj_replay)
>   File "/data/private/ljm/ManiSkill2-Learn/maniskill2_learn/apis/run_rl.py", line 293, in main_rl
>     train_rl(
>   File "/data/private/ljm/ManiSkill2-Learn/maniskill2_learn/apis/train_rl.py", line 336, in train_rl
>     training_infos = agent.update_parameters(replay, updates=total_updates)
>   File "/data/private/ljm/ManiSkill2-Learn/maniskill2_learn/methods/brl/bc.py", line 103, in update_parameters
>     [pred_dist, pred_action] = self.actor(sampled_batch["obs"], mode="dist_mean")
>   File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1111, in _call_impl
>     return forward_call(*input, **kwargs)
>   File "/opt/conda/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 952, in forward
>     if torch.is_grad_enabled() and self.reducer._rebuild_buckets():
> RuntimeError: Expected to have finished reduction in the prior iteration before starting a new one. This error indicates that your module has parameters that were not used in producing loss. You can enable unused parameter detection by passing the keyword argument `find_unused_parameters=True` to `torch.nn.parallel.DistributedDataParallel`, and by 
> making sure all `forward` function outputs participate in calculating loss. 
> If you already have done the above, then the distributed data parallel module wasn't able to locate the output tensors in the return value of your module's `forward` function. Please include the loss function and the structure of the return value of `forward` of your module when reporting this issue (e.g. list, dict, iterable).
> Parameter indices which did not receive grad for rank 1: 40
>  In addition, you can set the environment variable TORCH_DISTRIBUTED_DEBUG to either INFO or DETAIL to print out information about which particular parameters did not receive gradient on this rank as part of this error
> 
> /opt/conda/lib/python3.8/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 14 leaked shared_memory objects to clean up at shutdown
>   warnings.warn('resource_tracker: There appear to be %d '
> '''
> ...

xuanlinli17, 2023-07-13:
It's a bug in ManiSkill2-Learn due to the unused parameters in `TanhHead` in DDP. We'll fix this later.

For now you can either

- use GaussianHead in https://github.com/haosulab/ManiSkill2-Learn/blob/main/configs/brl/bc/pointnet_soft_body.py
- use a single gpu for BC training
- add `find_unused_parameters=True` in https://github.com/haosulab/ManiSkill2-Learn/blob/83dfe26c73b6ce6b0388a0fa07493f340e36dd44/maniskill2_learn/utils/torch/module_utils.py#L302

xuanlinli17, 2023-07-14:
Fixed ManiSkill2-Learn by adding `find_unused_parameters=True`

lijinming2018, 2023-07-14:
> Fixed ManiSkill2-Learn by adding `find_unused_parameters=True`

Thank you for your prompt reply

closed 2023-07-14 by xuanlinli17