mani-skill/ManiSkill #112: How to fix the height of a mobile manipulator?

songlin, 2023-07-14:
Hi,

When I run a random action in the `OpenCabinetDrawer-v1` env,  I observe that the mobile base's height changes. I even tried to disable the base control by setting `action[:4] = 0`, but the height still changed "randomly", which I think is due to the physics. So my question is how to fix the height of a mobile manipulator when controlling it?  I am asking this question because, in the real world, the base height is usually fixed.

```
env = gym.make("OpenCabinetDrawer-v1", obs_mode="rgbd", control_mode="base_pd_joint_vel_arm_pd_joint_vel")
env.seed(0)  # specify a seed for randomness
obs = env.reset()
done = False

while not done:
    action = env.action_space.sample()
    # Setting action[:4] = 0 here DOES NOT WORK!!
    obs, reward, done, info = env.step(action)
    env.render()  # a display is required to render
env.close()

```

Thank you.
Songlin

xuanlinli17, 2023-07-14:
Hi Songlin,

This is indeed due to physics. When the arm moves, the height will be affected as well.

To fix the mobile manipulator, you need to modify the robot urdf (for *single-arm* mobile panda, see https://github.com/haosulab/ManiSkill2/blob/fc08823bf96791946591a508f336d724fa7cad26/mani_skill2/assets/descriptions/mobile_panda_single_arm.urdf#L57). 

Then, define a custom panda variant in https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/agents/configs/panda/variants.py by overriding the urdf path. Define a custom panda robot in https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/agents/robots/mobile_panda.py as well.

Finally, define a custom environment by inheriting the original environment (https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/envs/ms1/open_cabinet_door_drawer.py), and override the agent loading functions (e.g., ` _load_agent`).

songlin, 2023-07-16:
@xuanlinli17 Thank you for the detailed instruction. I will try it later.

closed 2023-09-24 by xuanlinli17