mani-skill/ManiSkill #116: Segmentation Fault Error

Rose32, 2023-07-24:
I run into "Segmentation fault" issues for all the single file implementations for both Imitation Learning and Reinforcement Learning examples available in https://github.com/haosulab/ManiSkill2/tree/main/examples/tutorials/ . Similarly, I have the same "Segmentation fault" error when I run the scripts in https://github.com/haosulab/ManiSkill2-Learn/tree/main/scripts/example_training/pretrained_model. I use a bash script to run the python files in a singularity container that supports the maniskill2 environment. The container doesn't support visualization for the time being so I am just trying to run the example codes without any rendering. I am using the recommended runtime configurations for all these files with only the location of the python files and the gpu-ids modified.

For example when I use the dapg_pickcube_rgbd.sh script (in https://github.com/haosulab/ManiSkill2-Learn/tree/main/scripts/example_training/pretrained_model)  as it is, with some additional job submission scripts, I get the following errors:
For 2 GPUs and --gpu-ids 0 --sim-gpu-ids 1

`/var/spool/slurm/d/job1170327/slurm_script: line 25: 27020 Segmentation fault      singularity exec $HOME/maniskill2.simg python /home/rgurung/projects/maniskill/ManiSkill2/ManiSkill2-Learn/run_rl.py /home/rgurung/projects/maniskill/ManiSkill2/ManiSkill2-Learn/configs/mfrl/dapg/maniskill2_rgbd.py --work-dir ./logs/dapg_pickcube_rgbd --gpu-ids 0 --sim-gpu-ids 1 --cfg-options "env_cfg.env_name=PickCube-v0" "env_cfg.obs_mode=rgbd" "env_cfg.control_mode=pd_ee_delta_pose" "rollout_cfg.num_procs=16" "env_cfg.reward_mode=dense" "agent_cfg.demo_replay_cfg.buffer_filenames=../ManiSkill2/demos/rigid_body/PickCube-v0/trajectory.none.pd_ee_delta_pose_rgbd.h5" "eval_cfg.num=100" "eval_cfg.save_traj=False" "eval_cfg.save_video=True" "train_cfg.total_steps=25000000" "train_cfg.n_checkpoint=5000000"`

For 4 GPUs and --gpu-ids 0 1 --sim-gpu-ids 2 3

```
Traceback (most recent call last):
  File "/home/rgurung/projects/maniskill/ManiSkill2/ManiSkill2-Learn/run_rl.py", line 487, in <module>
    main()
  File "/home/rgurung/projects/maniskill/ManiSkill2/ManiSkill2-Learn/run_rl.py", line 450, in main
    mp.spawn(run_one_process, args=(world_size, args, cfg), nprocs=world_size, join=True)
  File "/opt/conda/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 239, in spawn
    return start_processes(fn, args, nprocs, join, daemon, start_method='spawn')
  File "/opt/conda/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 197, in start_processes
    while not context.join():
  File "/opt/conda/lib/python3.8/site-packages/torch/multiprocessing/spawn.py", line 140, in join
    raise ProcessExitedException(
torch.multiprocessing.spawn.ProcessExitedException: process 0 terminated with signal SIGSEGV
```

At this point I am not sure if there's a problem with the container I am using or if there's a problem with the code and the resources it requires. Would you have an idea as to what might be causing this problem and how to go about it? Please let me know if you require any additional information to diagnose the issue.

Thanks!
Rose

xuanlinli17, 2023-07-24:
If you simply run
```
import mani_skill2.envs, gym
env=gym.make('PickCube-v0', obs_mode='rgbd')
obs=env.reset()
```

Will there be an error?

Rose32, 2023-07-24:
Unfortunately, running those lines gives the same segmentation fault error. It fails on the second line too when I comment out the last line. The full error in the output file is:

```
[2023-07-24 16:45:11.433] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing
[2023-07-24 16:45:11.433] [svulkan2] [warning] Continue without GLFW.
[2023-07-24 16:45:11.448] [svulkan2] [error] Some required Vulkan extension is not present. You may not use the renderer to render, however, CPU resources will be still available.
/var/spool/slurm/d/job1170340/slurm_script: line 18: 29562 Segmentation fault      singularity exec $HOME/maniskill2.simg python test.py

```
The segmentation fault error is always preceded by the vulkan errors. I have been assuming these are separate issues but I am not sure. I have been ignoring those because I am aware the container I am using doesn't support that.

xuanlinli17, 2023-07-24:
Could you troubleshoot according to the instructions here? https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting

Rose32, 2023-07-24:
Do we need to have the Vulkan extension installed even though we aren't concerned with rendering for the time being? Are you suggesting that we need the Vulkan support even to just create a gym environment for maniskill2? I currently do not have the necessary access to do that in the HPC system I am working in. But if that is the case, I might have to find a way to do that.

xuanlinli17, 2023-07-24:
Not really, it's for debugging purposes.

Do the 3 json files in the troubleshooting section exists?

Rose32, 2023-07-27:
I just confirmed that we do have the 3 json files and that Vulkan is installed in the container. It has the libraries, and it produces the warnings that I posted when running the code saying it cannot do the rendering but other parts (CPU) should work. 

I added -q -X faulthandler to the python line and got a backtrace of the calls:

```
[2023-07-26 10:01:02.895] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing
[2023-07-26 10:01:02.895] [svulkan2] [warning] Continue without GLFW.
[2023-07-26 10:01:02.912] [svulkan2] [error] Some required Vulkan extension is not present. You may not use the renderer to render, however, CPU resources will be still available.
Fatal Python error: Segmentation fault

Current thread 0x00002ae3f43114c0 (most recent call first):
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/sensors/camera.py", line 143 in __init__
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 407 in _setup_cameras
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 360 in reconfigure
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 473 in reset
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 178 in __init__
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27 in __init__
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22 in __init__
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34 in make
  File "/opt/conda/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 92 in make
  File "/home/cousins/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 73 in make
  File "/home/cousins/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 106 in make
  File "/home/cousins/.local/lib/python3.8/site-packages/gym/envs/registration.py", line 184 in make
  File "bc_liftcube_state.py", line 143 in main
  File "bc_liftcube_state.py", line 265 in <module>
Segmentation fault
```

It is the "sensors.camera" call that is causing the problems in camera.py on Line 143, i.e.
 
             self.camera = scene.add_camera(

It there any work around for this?

xuanlinli17, 2023-07-27:
Could you update your nvidia driver to the latest like `nvidia-driver-530`?

Also what's your local ubuntu version and what happens when you run

```
sudo apt-get install vulkan-utils
vulkaninfo
```

Rose32, 2023-08-01:
This was solved by adding the --nv parameter in the singularity command. Thanks!

closed 2023-08-01 by Rose32