mani-skill/ManiSkill #117: VecEnv not works with

songlin, 2023-07-25:
Hi,

When I try to run a vectorized env of  `OpenCabinetDoor-v1` as following:
```
python mani_skill2/examples/demo_vec_env.py --env-id=OpenCabinetDoor-v1
```
I encountered following error:
```
2023-07-25 16:04:26,943 - mani_skill2 - ERROR - 'NoneType' object has no attribute 'vertices'
Traceback (most recent call last):
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/vector/vec_env.py", line 56, in _worker
    env = env_fn()
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/vector/registration.py", line 11, in _make_env
    env = env_spec.make(**kwargs)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/open_cabinet_door_drawer.py", line 39, in __init__
    super().__init__(*args, **kwargs)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/base_env.py", line 56, in __init__
    super().__init__(*args, **kwargs)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 178, in __init__
    obs = self.reset(reconfigure=True)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/open_cabinet_door_drawer.py", line 163, in reset
    return super().reset(seed=seed, reconfigure=reconfigure, model_id=model_id)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/base_env.py", line 88, in reset
    ret = super().reset(seed=self._episode_seed, reconfigure=reconfigure)
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 473, in reset
    self.reconfigure()
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/sapien_env.py", line 359, in reconfigure
    self._load_articulations()
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/open_cabinet_door_drawer.py", line 67, in _load_articulations
    self._set_cabinet_handles_mesh()
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/envs/ms1/open_cabinet_door_drawer.py", line 95, in _set_cabinet_handles_mesh
    meshes.extend(get_visual_body_meshes(visual_body))
  File "/home/songlin/miniconda3/envs/sapien/lib/python3.8/site-packages/mani_skill2/utils/trimesh_utils.py", line 42, in get_visual_body_meshes
    vertices = render_shape.mesh.vertices * visual_body.scale  # [n, 3]
AttributeError: 'NoneType' object has no attribute 'vertices'
```

Thanks for the help.

StoneT2000, 2023-07-25:
This issue is caused by using dense reward and vec env for some environments. For OpenCabinetDoor-v1, the dense reward calculation uses the visual body meshes to get a fairly good reward function. However, this is incompatible with the vec env as the vec env uses a separate render process for visual data processing.

Currently we do not have a easy solution for this other than to recommend designing your own dense reward that does not use the original dense reward.

songlin, 2023-07-25:
I see. Thank you.

closed 2023-07-25 by songlin