mani-skill/ManiSkill #118: [Question] Configuring for state-based PPO on PickCube

yichao-liang, 2023-07-25:
Hi, following up on the previous discussion, I was wondering if I can seek some advices regarding an optimal configuration for training state-based PPO on the PickCube task, as outlined in the [tutorials](https://github.com/haosulab/ManiSkill2/blob/main/examples/tutorials/reinforcement-learning/sb3_ppo_liftcube_state.py).

I have executed the code using 32 environments running in parallel. However, the resulting performance has been somewhat unpredictable, as illustrated in the attached Tensorboard screenshot:

[image]

In some cases, the model performs successfully, evidenced by the blue curve in the screenshot. However, in a majority of instances, there is a sudden performance drop following an initial increase, like the model with grey curve.

Given that I am relatively new to training RL models, I would greatly appreciate any insights you might have to improve stability in the training process. Do you recommend specific adjustments, such as reducing the learning rate or implementing a learning rate scheduler, to address this issue?

I am grateful for your guidance and look forward to your valued feedback.

xuanlinli17, 2023-07-25:
In your case, `n_steps=rollout_steps // num_envs`, and since `rollout_steps=3200`, `num_envs=32`, `n_steps=100`. However, the environment horizon for pickcube is 200, so you are resetting the environment before an episode is completed in policy rollout, which harms sample efficiency. When you are using a larger `num_envs`, please set `rollout_steps` higher, such as 6400 or 9600.

yichao-liang, 2023-07-25:
Many thanks for your prompt reply! 

By the environment horizon, are you referring to the `max_episode_steps=200` in the decorator of the environment class?  And you're suggesting that the `n_steps` should be larger or equal to `max_episode_steps`? Is it not the case that if `n_steps` is smaller than `max_episode_steps`, when `collect_rollouts` is called, the data collection would continue from the same environments where the previous collection has not reached the `max_episode_steps` limit, which wouldn't damage the performance?

On another note, it seems the maximum episode steps is overridden by argparse and set to 100 as per this [link](https://github.com/haosulab/ManiSkill2/blob/fc08823bf96791946591a508f336d724fa7cad26/examples/tutorials/reinforcement-learning/sb3_ppo_liftcube_state.py#L67). This should overwrite the `max_episode_steps=200` for PickCube, right?

Considering these aspects, do you advise altering the `args.max_episode_steps` to 200 as well?

Thank you again for your valuable insights and guidance.

xuanlinli17, 2023-07-25:
Yes, `max_episode_steps=200` in the decorator of the environment class equals environment horizon. In the tutorial, the max episode steps is indeed overridden to 100. Since you have 32 parallel environments in the tutorial, the effective `n_steps` for each environment is 3200(rollout_steps)/32(num_envs)=100.

In fact, when you use 32 parallel environments, even though `n_steps` per environment equals 100, which is the `max_episode_steps` in the tutorial, it is still too small. Actually, I'd recommend `n_steps` per environment be **a lot larger** than `max_episode_steps`. Otherwise,  when the agent starts to succeed, partial trajectories will occur, and will cause ambiguities in the trajectory return, which causes large value estimation errors and account for the sudden drop in performance. (e.g., when `n_steps=100`, assuming one successful trajectory has length 55 and another successful one has length 45 but truncated, then the advantages calculated for the untruncated second trajectory vs. the advantage calculated for untruncated second trajectory will be significantly different from each other. This will also cause ambiguities in the value (V) loss in PPO)

Thus I encourage you to set a much higher `rollout_steps` to something like `9600` or `12800`, or reduce `n_envs`, to ensure that the `n_steps` for each environment is large enough.

Or, there is another solution: when collecting policy rollouts, don't reset the environment until `Timelimit.Truncated` (environment reaches `max_episode_steps`) by ignoring the success signal. You can see [here](https://github.com/haosulab/ManiSkill2-Learn/blob/c3a4d9d6c059ed301629a8cb230d9fa0f38f6ce9/maniskill2_learn/env/wrappers.py#L228) for an example.

StoneT2000, 2023-07-26:
Small error here but the tutorials use a continuous task representation, so train episodes should not end early. Max episode steps is set to 100 instead of the default 200 is a simple trick to improve sample efficiency (does not always work for all envs) as it only really takes 50-60 steps to actually solve PickCube.

However Xuanlin is right, you should probably set the rollout steps higher regardless given you are using 32 parallel environments.

yichao-liang, 2023-07-26:
Thanks, @xuanlinli17 and @StoneT2000, for providing the help.

@xuanlinli17, to make sure I understand you correctly, by 
> the advantages calculated for the untruncated second trajectory vs. the advantage calculated for untruncated second trajectory will be significantly different from each other in the GAE value estimator.

you mean that, say, when `n_steps=max_episode_steps=100`, assume the agent has finished the first successful trajectory with 55 steps starting from `t=0` and will finish the second one in 50 steps. But due to `n_steps=100`, the second one is truncated at step 45.  This truncation results in a substantial discrepancy compared to if the trajectory were allowed to complete at the 50th step.

For context, I have conducted two experiments with parameters max_episode_steps=200, rollout_steps=9600, num_envs=32, and thus n_steps=300. Nevertheless, the issue persists, as evidenced by the attached screenshot.

[image]

@StoneT2000 Thanks for highlighting the function `ContinuousTaskWrapper`. From your explanation, it seems that training should not conclude upon successful task completion, but only upon reaching max_episode_steps.

I will adjust the batch size and the number of update epoch to see if they help.

Thank you once again for your invaluable support and guidance.

StoneT2000, 2023-07-26:
It seems your policy has already done fine? And then collapsed later. This is some training instability, perhaps either increase batch size, lower the target KL.

yichao-liang, 2023-07-28:
Thanks, @StoneT2000, I appreciate your insightful suggestions! They've been immensely helpful in guiding my experiments.

Starting from the default model (in blue curve, below) with 
`n_envs=32, n_steps=100, batch_size=400, target_kl=0.05, n_epochs=15, clip_range=0.2`

I experimented with the model in green and red with the following configurations, respectively:
green: `n_envs=32, n_steps=100, batch_size=3200, target_kl=0.02, n_epochs=10, clip_range=0.2`
red.   : `n_envs=32, n_steps=100, batch_size=3200, target_kl=0.01, n_epochs=10, clip_range=0.1`

Both configurations showed increased stability compared to the default (blue curve) model, but they both unfortunately still collapsed after approximately 60M steps.

[image]

Given these observations, I'm keen on hearing your thoughts on potential next steps.

I'm concerned that continuing to decrease target_kl or clip_range might hinder learning effectiveness. However, I wonder if introducing a clip range or learning rate scheduler could be beneficial?

Thank you for your attention!

StoneT2000, 2023-07-29:
Maybe try n_steps=200 or 400, and play around with target kl, clip frac, n_epochs

Increasing n steps increases the number of episodes before updating, making it less likely to have a bad batch of episodes or learn something bad.

yichao-liang, 2023-07-29:
Thanks for the suggestions. Sorry, there was a typo in the previous comment—I was actually already using `n_steps=400` with the model in with the green and red curve.

yichao-liang, 2023-07-30:
Hi, thanks so again for all your support on this question. I've finally stabilized the training by using 
`n_envs=32, batch_size=3200, kl_target=0.02, n_epochs=10, n_rollout_steps400` and linear learn_rate_scheduler and clip_range_scheduler that anneal learning rate and clip range from 3e-4 to 1e-5 and from 0.1 to 0.05 in 20M steps. The training has been fairly stable for 250M steps now.

[image]

closed 2023-07-30 by yichao-liang

StoneT2000, 2023-07-31:
Huh so it seems like n_steps did the trick. I'm curious if you don't mind could you try n_envs=16 and n_steps=200? Everything else the same. Maybe this is some SB3 bug

yichao-liang, 2023-07-31:
It seems it's the learning rate and clip range scheduler that did the trick, because when I used `n_steps` without the scheduler [earlier](https://github.com/haosulab/ManiSkill2/issues/118#issuecomment-1656704920), the instability also appeared.