mani-skill/ManiSkill #122: How to set large ee_target_pose controllers?

HeegerGao, 2023-07-31:
Hi, ManiSkill2 is such amazing work! I want to use it for robotic grasping research, where at each timestep the agent outputs the ee_target_pose in (probably any reachable pose) the scene. However, I find that in ManiSkill2, the ee_target_pose controller limits every dimension of the target pose output to a range of [-1.0, 1.0], which limits the action space. I want to ask how should I change the target_ee_pose controller (probably [here](https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/agents/controllers/pd_ee_pose.py#L123)?) to expand its output limits to any reachable target pose. Do I just need to modify [this class](https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/agents/controllers/pd_ee_pose.py#L123)? Or do I still need to modify some other configurations (e.g., the control frequency or something caused by longer action time from farther target pose)? Thanks!

xuanlinli17, 2023-07-31:
As a background, different pd ee pose controllers' implementation follows from `mani_skill2/agents/controllers/pd_ee_pose.py`

The action range being [-1.0, 1.0] is caused by the `normalize_action=True` configuration. In this case, there are `pos_lower` and `pos_upper` that control the position range (action = -1 maps to `pos_lower` and action = 1 maps to `pos_upper`; these translation actions are specified by `action[0:3]` in the `pd_ee_delta_pose` controller)

For rotation, the `rot_bound` controls the maximum angle of rotation. The rotation is represented in the axis-angle format, so if `rot_bound=0.1`, then the rotation is at most 0.1 rad with respect to the axis specified by `normalized(action[3:6])` in the `pd_ee_delta_pose` controller.

To allow the end effector to reach any pose you desire, you can create a new controller configuration by setting `normalize_action=False`, as follows:
```
arm_pd_ee_pose = PDEEPoseControllerConfig(
            self.arm_joint_names,
            -10.0,
            10.0,
            2*np.pi,
            self.arm_stiffness,
            self.arm_damping,
            self.arm_force_limit,
            ee_link=self.ee_link_name,
            use_delta=False,
            use_target=False,
            normalize_action=False,
            frame='base', # such that the position & rotation actions are always with respect to the base frame, since ee frame can change as control is being executed
        )
```
In this case the actions will not be normalized and will represent the absolute position & rotation
(Note that since it's a pd controller, it will take some time to reach the target pose after a control command is sent; so you can set `control_freq` to be a lower number in the environment)

Then, add this controller config to the robot's `controller_configs` under the robot's `defaults.py`, see
https://github.com/haosulab/ManiSkill2/blob/fc08823bf96791946591a508f336d724fa7cad26/mani_skill2/agents/configs/panda/defaults.py#L83
https://github.com/haosulab/ManiSkill2/blob/fc08823bf96791946591a508f336d724fa7cad26/mani_skill2/agents/configs/panda/defaults.py#L146

HeegerGao, 2023-08-01:
Thanks very much for your quick reply! It works.

closed 2023-08-01 by HeegerGao

TongZhangTHU, 2023-12-15:
Hello @xuanlinli17,

Thanks for your explanation. Following your advice, I've incorporated both "arm_pd_ee_pose" and "pd_ee_pose" into [ManiSkill2/mani_skill2/agents/configs/panda/defaults.py](https://github.com/haosulab/ManiSkill2/blob/fc08823bf96791946591a508f336d724fa7cad26/mani_skill2/agents/configs/panda/defaults.py). However, I've encountered an issue when using the mani_skill2.trajectory.replay_trajectory function in the "pd_ee_pose" control mode.

Specifically, an AssertionError is triggered at [this line in replay_trajectory.py](https://github.com/haosulab/ManiSkill2/blob/main/mani_skill2/trajectory/replay_trajectory.py#Line110). I'm curious if this assertion is an intended part of the functionality, and if so, how we should properly utilize replay_trajectory with the "pd_ee_pose" control mode to avoid this error.

Any insights or further guidance you could provide on this matter would be greatly appreciated.

Thank you for your time and assistance.