mani-skill/ManiSkill #124: Change of camera configurations after vectorization

Yefanov, 2023-08-01:
I encountered a strange behavior with the `TurnFaucet-v0` environment. When I wrap it in a vector environment, the camera configurations seem to change, resulting in significantly different images from the base and hand cameras (either the camera is too close to the faucet or the faucet appears much larger than the robot).

Here's the code to recreate the issue:

```python
import matplotlib.pyplot as plt
from mani_skill2.vector import make

# Create the vectorized environment
venv = make('TurnFaucet-v0', 1, obs_mode='rgbd', reward_mode='sparse')

# Reset the environment and get the initial observation
obs = venv.reset()

# Close the environment
venv.close()

# Display the image from the base camera
plt.imshow(obs['image']['base_camera']['rgb'].cpu().numpy()[1])
```
[image]

Upon investigating the `mani_skill2/vector/registration.py` file, I discovered that the `env_fn` function generates the environment with correct images. However, both `RGBDVecEnv` and `VecEnv` produce environments with distorted ones.

Interestingly, the `LiftCube-v0` environment works perfectly fine.

Please note that to run the vectorized `TurnFaucet-v0` environment, you need to use the 0.5.0 branch or visit this GitHub repository: https://github.com/Yefanov/mani-skill2/tree/fix_vec_env

StoneT2000, 2023-08-02:
Yes i think we also see this issue on our draft 0.5.0 update, we are investigating it now, thanks for bringing it up!

StoneT2000, 2023-08-03:
Yes I think this is the issue I have with turn faucet, trying to fix now in 0.5.0

StoneT2000, 2023-08-03:
Ok the issue is with vecenv where the worker does not receive the reset command to reset and correct model size. Not sure why just yet

StoneT2000, 2023-08-03:
Actually it looks like this bug also surfaced up in the previous version? Perhaps this is a larger bug than expected.

Yefanov, 2023-08-03:
Hello, I have tested with most environments:

Based on 0.4.0:
- LiftCube, PickSingleYCB, PickClutterYCB, PlugCharger, PegInsertionSide: they look great.
- PickSingleEGAD: The EGAD object appears enlarged.
- AssemblingKits: The size is preserved, but the colors of the kits are lost, and strange walls are added to the scene.

Based on 0.4.0 (actually, I tested it only with a minor fix that was pulled into 0.5.0, so the problem is not in the 0.5.0 update):
- OpenCabinetDoor, OpenCabinetDrawer, PushChair: they look like the whole scene became larger or the camera closer. However, the relative sizes between the robot and objects are preserved.
- TurnFaucet, MoveBucket: The faucet and bucket are enlarged.

Here is some examples:
[image]
[image]

Here is comparison between before and after vectorization for OpenCabinetDrawer:
[image]
[image]

StoneT2000, 2023-08-03:
We've identified that there is a issue with sapien actually. Thanks for helping find all the scaling issues! We're fixing it soon

StoneT2000, 2023-08-03:
@fbxiang just published a fix that should work on your code (and on our new 0.5.0 branch). Can you run `pip install sapien==2.2.2`?

Yefanov, 2023-08-04:
Thank you! The assets for each environment are of the correct sizes. There is one small bug, which is preserved, however, where the kits from AssemblingKits appear grayish instead of being colorful.

closed 2023-08-04 by Yefanov

StoneT2000, 2023-08-04:
What do you mean by grayish?

Yefanov, 2023-08-07:
Apologies for any confusion. What I meant to convey is that the assets appear grey instead of being colorful.
[image]