mani-skill/ManiSkill #129: Make a proper test for trajectory replay tool

StoneT2000, 2023-08-04:
Currently we just run the replay tool for coverage purposes. Should add tests to check the actual contents of the saved trajectories/videos.

zhuoqun-chen, 2023-12-28:
@StoneT2000 Agree! Also, is there a way to save successful trajectories produced during policy evaluation so they may be used for imitation learning?

StoneT2000, 2023-12-28:
I recommend using the RecordEpisodeWrapper from the maniskill2 package which will automatically save trajectories you run through the env following the maniskill2 format

StoneT2000, 2023-12-28:
mani_skill2/utils/wrappers/record.py
this one

zhuoqun-chen, 2023-12-29:
@StoneT2000 Hi Stone! I tried this one, but it seems that the saved `xxx.h5` is not readable by using:

```python
import h5py

f = h5py.File("path_to_trajectory_file/xxx.h5", "r")
```

the corresponding error messages are:

```
Traceback (most recent call last):
  File "/home/zqchen/phd/gsl/logs/save_traj_only/videos/read_data.py", line 3, in <module>
    f = h5py.File("/home/zqchen/phd/gsl/logs/save_traj_only/videos/20231228_152428.h5", "r")
  File "/home/zqchen/miniconda3/envs/gsl/lib/python3.8/site-packages/h5py/_hl/files.py", line 562, in __init__
    fid = make_fid(name, mode, userblock_size, fapl, fcpl, swmr=swmr)
  File "/home/zqchen/miniconda3/envs/gsl/lib/python3.8/site-packages/h5py/_hl/files.py", line 235, in make_fid
    fid = h5f.open(name, flags, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 102, in h5py.h5f.open
OSError: Unable to synchronously open file (bad object header version number)
```

A potential reason is that I keyboard interrupted the training (without rendering window) so evaluation environment's `close` method is not called so `self._h5_file.close()` is not called?

StoneT2000, 2024-01-05:
That is also possible, can you try closing it first before using it?

closed 2024-11-08 by StoneT2000