mani-skill/ManiSkill #130: Pixels shift occur in the vector environment when using smaller image width for camera

Yefanov, 2023-08-04:
Hello, I have encountered this issue: pixels shift occur in the vector environment when the camera configurations are modified with a smaller image width.

```python
from mani_skill2.vector import make
import matplotlib.pyplot as plt

venv = make('LiftCube-v0', 1, obs_mode='rgbd', reward_mode='sparse', camera_cfgs={'base_camera': {'width': 97}})
obs = venv.reset()
venv.close()

plt.imshow(obs['image']['base_camera']['rgb'].cpu().numpy()[0])
```
[image]

StoneT2000, 2023-08-04:
Thanks for bringing it up, we will take a look.

Curious though why did you try width of 97?

Yefanov, 2023-08-07:
I just picked a random prime number. Initially, I worked with ms1 environments and modified the camera configurations from robot-based 160x400 to static 128x128. I chose a prime number because using 128x64 would result in a squeezed image. I wanted to avoid any common divisors that could lead to incorrect assumptions.

StoneT2000, 2023-08-08:
Looks like you are quite spot on. We aren't sure just yet what the bug is but a temporary solution is to ensure the camera sizes for both hand and base cameras are the same. That seems to fix the bug.

StoneT2000, 2023-08-08:
Actually we know what the bug is. The bug is at the ManiSkill2 level. What happens is the vec env observations are padded because one camera is smaller. That's where the black part comes in. The RGBDVecEnv wrapper does not handle the padding quite correctly.

If you reallly want to use a prime number width, check the RGBDVecEnv wrapper and reshape it correctly. Otherwise just use nicer camera sizes.

closed 2023-08-08 by StoneT2000

Yefanov, 2023-08-08:
This "fix" works for me (line 247 of vec_env.py):

```python
tensor = tensor[:, :w].flatten().reshape((-1, h, w, 4))
```

However, this will fail when used with an image height smaller than the default.

The issue is that padding always gets added to the bottom of an image. But when using a smaller width, padding should be added to the right side of the image.

In any case, using the same camera sizes is the solution for me. Thank you!