mani-skill/ManiSkill #133: Leak of GPU memory when using set_environment_map_from_files function

lyshuga, 2023-08-10:
Hi,

We have observed one issue. After every reset it was applied "set_environment_map_from_files" to generate a new skybox, but unfortunately after every 10-20 reset the amount of GPU memory used increased by 100-200 MB. So, after let's say 100 resets we run out of GPU memory. The function "set_environment_map_from_files" is the wrapper for C++ function, I was wondering if it possible to somehow overcome this issue.

Thanks

xuanlinli17, 2023-08-10:
Looks like the memory consumption is due to the cached resources. You can release the memory by

`env.unwrapped._renderer.clear_cached_resources()`

StoneT2000, 2024-04-30:
Closing this issue as it has gone stale. Feel free to reopen if not.

closed 2024-04-30 by StoneT2000

IrvingF7, 2025-04-24:
> Looks like the memory consumption is due to the cached resources. You can release the memory by
> 
> `env.unwrapped._renderer.clear_cached_resources()`

@StoneT2000 @xuanlinli17 
Hi, Sorry to dig up this old issue. I am facing something similar in that, whenever I make a new environment in a single process, I notice that my VRAM goes up, and it never comes down. Reset doesn't increase VRAM, but making new env does. 

I tried this api but apparently it was not in the codebase anymore. Is there something similar that I can do? If you think this is an entirely different issues, I am more than happy to open a new independent issue.

Thanks in advance

StoneT2000, 2025-05-02:
Will address this in the simpler-env repo