mani-skill/ManiSkill #140: http 404 when running `python -m mani_skill2.utils.download_demo PickSingleYCB-v0`

mlzxy, 2023-08-23:
Error like this `urllib.error.HTTPError: HTTP Error 404: Not Found`. 

I verify this error on both local machine and remote server.

xuanlinli17, 2023-08-23:
It works fine for me. Perhaps you need vpn to access google since the demos are stored on google drive?

mlzxy, 2023-08-23:
Thank you for your reply. I am downloading the file overseas actually. 

 

I can download other trajectories but not this. Could you check huggingface again like here https://huggingface.co/datasets/haosulab/ManiSkill2/tree/main/demos/v0/rigid_body/PickSingleYCB-v0?  It contains a zip file instead of trajectory h5 and json.

StoneT2000, 2023-08-23:
Fixing the HF link now

StoneT2000, 2023-08-23:
While we are waiting on a review and releasing the new pypi package to fix the bug, you may download the dataset yourself here: https://huggingface.co/datasets/haosulab/ManiSkill2/blob/main/demos/v0/rigid_body/PickSingleYCB-v0/PickSingleYCB-v0.zip

And unzip it to `demos/v0/rigid_body`

mlzxy, 2023-08-23:
> While we are waiting on a review and releasing the new pypi package to fix the bug, you may download the dataset yourself here: https://huggingface.co/datasets/haosulab/ManiSkill2/blob/main/demos/v0/rigid_body/PickSingleYCB-v0/PickSingleYCB-v0.zip
> 
> And unzip it to `demos/v0/rigid_body`

Thanks for your guidance. Will do.

StoneT2000, 2023-08-23:
closed with latest release 0.5.1

commit 535d85a2 (2023-08-23): Fix #140
closed 2023-08-23 by StoneT2000