mani-skill/ManiSkill #143: Would the transporter network baseline get released?

mlzxy, 2023-08-25:
Hi, 

It was mentioned in the paper that transporter network is tested on the AssemblingKits task (table 5). I wonder whether the corresponding implementation will be released. 

Thanks!

StoneT2000, 2023-08-26:
We were able to release the TPN baseline here https://github.com/haosulab/ManiSkill2-SPA

The results don't match the paper completely due to changes in the benchmark and envs but it should still work. Note that this baseline is a very "minimal" baseline, there is a lot of room for tuning especially given that ManiSkill2 assembling kit environment is more realistic and strict than what TPN tests on.

mlzxy, 2023-08-26:
Go it. Thanks for your help!

closed 2023-08-26 by StoneT2000