mani-skill/ManiSkill #147: [svulkan2] [error] Some required Vulkan extension is not present. You may no t use the renderer to render, however, CPU resources will be still available.

Charlie0257, 2023-09-05:
When I try 20 parallel envs or even 1 env, there is the error occurinng——

`[2023-09-04 10:25:11.337] [svulkan2] [error] Some required Vulkan extension is not present. You may not use the renderer to render, however, CPU resources will be still available.`

`Use trivial renderer without color. /home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/site-packages/gym/core.py:317: DeprecationWarning: WARN: Initializing wrapper in old step API which returns one bool instead of two. It is recommended to set new_step_api=Trueto use new step API.`

`This will be the default behaviour in future. deprecation( current task is: laptop /home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/site-packages/sapien/core/renderer_config.py:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html import pkg_resources /home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to pkg_resources.declare_namespace('mpl_toolkits'). Implementing implicit namespace packages (as specified in PEP 420) is preferred to pkg_resources.declare_namespace. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages declare_namespace(pkg) /home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to pkg_resources.declare_namespace('sphinxcontrib'). Implementing implicit namespace packages (as specified in PEP 420) is preferred to pkg_resources.declare_namespace. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages declare_namespace(pkg)`

`Traceback (most recent call last): File "train_agent.py", line 309, in <module> envs = SubprocVecEnv([create_env_fn] * args.procs, "spawn")  # train on a list of envs. File "/home/hao_zhang0257/Projects/TALD/stable_baselines3/common/vec_env/subproc_vec_env.py", line 109, in __init__ self.remotes[0].send(("get_spaces", None)) File "/home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/multiprocessing/connection.py", line 206, in send self._send_bytes(_ForkingPickler.dumps(obj)) File "/home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/multiprocessing/connection.py", line 411, in _send_bytes self._send(header + buf) File "/home/hao_zhang0257/anaconda3/envs/TALD/lib/python3.8/multiprocessing/connection.py", line 368, in _send n = write(self._handle, buf) BrokenPipeError: [Errno 32] Broken pipe`

Thanks for any help! :) @Daniel-Liu-c0deb0t @tongzhoumu @xiqiangliu

xuanlinli17, 2023-09-05:
Please see the troubleshooting section of https://haosulab.github.io/ManiSkill2/getting_started/installation.html

Charlie0257, 2023-09-05:
Thanks for the patient reply. @xuanlinli17 

I will try the possible solution and reply soon. :)

Charlie0257, 2023-09-06:
Thanks for the help! :)
I created the nvidia_icd.json under /usr/share/vulkan/icd.d/, and it worked! :)

I will close this issue.

closed 2023-09-06 by Charlie0257