mani-skill/ManiSkill #150: Class attribute not being updated after setting it manually

yichao-liang, 2023-09-15:
I have a question about having a step counter in for vectorized environment. It seems that the counter fails to be updated after the variable is set manually.

The problem can be illustrated by the minimal example below. To walk through it, I create a `PickCubeEnvMod` class which has an additional attribute that counts how many times it has stepped. Then I create a vectorized environment `env`. After running it for 10 steps, `step_counter` now has value 10, as expected. But when I run it again after setting it to 100, I got 100 after the run when I'm expecting 110. 

I want to set `step_counter` manually as, e.g., when I continue training from a previously save model, I want it to resume from the state when the previous model is saved, so that an environment parameter scheduler in my environment can work properly.

I realize that this might not be a problem with maniskill2 in particular. But any suggestions on how to resolve this is greatly appreciated!
```
import numpy as np
import gymnasium as gym

from mani_skill2.utils.registration import register_env
from stable_baselines3.common.vec_env import SubprocVecEnv
from mani_skill2.envs.pick_and_place.pick_cube import PickCubeEnv

@register_env("PickCube-v2", max_episode_steps=200)
class PickCubeEnvMod(PickCubeEnv):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self.step_counter = 0

    def step(self, action):
        self.step_counter += 1
        return super().step(action)

def main():
    def make_env(
        env_id: str,
        max_episode_steps: int = None,
        record_dir: str = None,
    ):
        def _init() -> gym.Env:
            # NOTE: Import envs here so that they are registered with gym in subprocesses
            import mani_skill2.envs

            env = gym.make(
                env_id,
                obs_mode="state",
                reward_mode="normalized_dense",
                control_mode="pd_ee_delta_pose",
                render_mode="cameras",
                max_episode_steps=max_episode_steps,
            )
            return env
        return _init

    env = SubprocVecEnv([make_env('PickCube-v2', max_episode_steps=100)
                        for _ in range(2)])
    env.reset()

    for _ in range(10):
        action = env.action_space.sample()
        action = np.repeat(action[None, :], 2, axis=0)
        env.step(action)

    # check the step counter, we should expect 10
    # we do get 10
    print(env.get_attr('step_counter'))

    # manually set the counter
    env.set_attr('step_counter', 100)
    for _ in range(10):
        action = env.action_space.sample()
        action = np.repeat(action[None, :], 2, axis=0)
        env.step(action)

    # check the step counter again, we should expect 110
    # but we still get 100
    print(env.get_attr('step_counter'))

if __name__ == "__main__":
    main()
```

StoneT2000, 2023-09-15:
could you try adding a print statement inside PickCubeEnvMod's step function and print the step counter? What do you get?

yichao-liang, 2023-09-15:
Thanks for your help! I found adding a setter method for the environment and use it to set the step_counter resolves this issue. :) 
```
@register_env("PickCube-v2", max_episode_steps=200)
class PickCubeEnvMod(PickCubeEnv):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self.step_counter = 0

    def set_step_counter(self, n):
        self.step_counter = n

    def step(self, action):
        self.step_counter += 1
        return super().step(action)
       
    # manually set the counter
    env.env_method('set_step_counter', 100) 
```

StoneT2000, 2023-09-15:
This is then probably a stable baselines bug, perhaps post it there, not sure what this is. 
Nice to see this soluation works.

closed 2023-09-15 by yichao-liang