mani-skill/ManiSkill #153: Cloth manipulation

YunxiTang, 2023-09-27:
Dear team,
Thanks for this nice project. I am wondering is there any support for cloth-like deformable object manipulation at this moment?
Thanks

StoneT2000, 2023-09-27:
I believe this is currently in progress, I'm not sure when we might release some cloth environments, @fbxiang ?

fbxiang, 2023-09-27:
Yes, this is in progress, but we do not have a release timeline just yet.

closed 2025-02-04 by StoneT2000

lhy0807, 2025-03-14:
I just wanted to follow up on this. Has the cloth manipulation env implemented yet?

prasuchit, 2025-05-03:
Hi @StoneT2000 , I'm looking to use soft bodies in my research. Is this feature available now? Is there an example I can follow?

StoneT2000, 2025-05-04:
Only partially in ManiSkill2, but that API is not easy to extend yourself just yet. This is currently being worked on now actually, but takes some time to make it clean and usable. I hope to get a beta version of the mpm system back up on our nightly package by July (will be based on our old mpm code + rewarped

I will point out, we aren't making soft body sims any better really apart from being one of the few sims maybe to support GPU parallelized sim+rendering of soft body sim. Issues like actual robot grasping of cloth that other sims have will also occur here since we are just making a simple decoupled mpm system (no unified rigid body soft body equations).

The soft body sim feature stuff is tracked in https://github.com/haosulab/ManiSkill/issues/729

prasuchit, 2025-05-04:
@StoneT2000 , thank you for the prompt response. I'm currently using [this repo](https://maniskill.readthedocs.io/en/latest/index.html) (just checking to make sure I'm using the right link). I've briefly used Isaac Sim, Mujoco, and more extensively Gazebo, Pybullet, etc. By far, ManiSkill has been the easiest to set up and add custom models to. I wasted almost two full days trying to get my custom models working in Robosuite and failed.

So I'm already hooked to ManiSkill. Any level of deformable object simulation it provides, I will take it. Other simulators like Isaac and Drake promise more realistic material modelling, but their docs are super confusing.

Thank you for creating, documenting, and maintaining ManiSkill so well. I'm sure it's going to become one of the most sought after simulators soon.

StoneT2000, 2025-05-04:
That documentation is the correct one for ManiSkill3, which is primarily just rigid body simulation.

I'm adding some soft body support in a decoupled manner to our physx backend to ManiSkill3

There is ManiSkill2 which is the 0.5.3 pypi package version which has some soft body support but quality wise and documentation wise no where near as good or simple as maniskill 3 right now.

And thanks for the kind feedback! Really great to hear our system and api design suits your needs!