mani-skill/ManiSkill #158: Template File for creating custom environments

StoneT2000, 2023-10-11:


CyanWatts, 2023-11-17:
I met similar problem too. Recently I wanted to build a customized kitchen scene, but I found the current customize_environments tutorial (https://github.com/haosulab/ManiSkill2/blob/main/examples/tutorials/customize_environments.ipynb) way too simple to help users fully build a custom environment. I'll appreciate it if anyone provides more detailed guide on environments customization (i.e., more information about static scenes, what class and function to implement etc.).

StoneT2000, 2023-11-17:
this will be part of an upcoming release that we are working on to make it easy to import scenes, build scenes, and build tasks, covering the main aspects of task creation: objects, scene setup, and task dynamics. Stay tuned!

CyanWatts, 2023-11-18:
> this will be part of an upcoming release that we are working on to make it easy to import scenes, build scenes, and build tasks, covering the main aspects of task creation: objects, scene setup, and task dynamics. We have some private beta work going on right now that has done some of this, I'll see which parts of it we can release publicly earlier
> 
> Stay tuned!

thanks for the reply! We’ll keep an eye on it

StoneT2000, 2023-12-08:
https://github.com/haosulab/ManiSkill2/pull/175 will include some new features for building more complex scenes + importing existing ones from some other datasets (althugh other datasets currently do not have fully articulated objects like cabinets/fridges). should be merged soon-ish

StoneT2000, 2023-12-14:
Ok all the updates are merged to the dev branch, the pre-release version is 0.6.0.dev0 so to use it i recommend using a new environment and running `pip install mani_skill2==0.6.0.dev0`

See the file in dev branch at `mani_skill2/envs/template.py`

closed 2024-04-30 by StoneT2000