mani-skill/ManiSkill #159: ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object

zyzshixiaobendan, 2023-10-17:
I ran into this problem when I was running：

File "/home/cy/PycharmProjects/tactile-vision-fusion/envs/continuous_insertion.py", line 395, in __initialize__
    renderer = sapien.VulkanRenderer()
RuntimeError: vk::Instance::enumeratePhysicalDevices: ErrorInitializationFailed

Then I typed in vulkaninfo to see if vulkan would work, and it reported an error

ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object 

ubuntu22.04

zyzshixiaobendan, 2023-10-17:
I am using nvidia 535.

zyzshixiaobendan, 2023-10-17:
I am using nvidia 535 and don't know what to do with it, help!

xuanlinli17, 2023-10-17:
Hi, please follow https://haosulab.github.io/ManiSkill2/getting_started/installation.html#troubleshooting  for debugging. You need to ensure the 3 json files exist.

zyzshixiaobendan, 2023-10-17:
Thank you for your reply! If the three json files are set up will the file hold error also disappear?

zyzshixiaobendan, 2023-10-17:
Thank you for your reply! If the three json files are set up will the file hold error also disappear? @xuanlinli17

xuanlinli17, 2023-10-17:
The error will most likely gone.

zyzshixiaobendan, 2023-10-18:
When I execute 'sudo apt-get install libglvnd-dev,' it doesn't generate the '10_nvidia.json' file. How can I resolve this? @xuanlinli17

zyzshixiaobendan, 2023-10-18:
You should reinstall vulkan if you get RuntimeError: vk::Instance::enumeratePhysicalDevices: ErrorInitializationFailed, and I've already reinstalled it and it still gives me the same error. I typed vulkaninfo into the terminal to see if vulkan could run properly and he reported ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object file: No such file or No such file or directory. Is the essential problem that this file is missing? Please help me to solve it, thanks! @xuanlinli17

StoneT2000, 2024-04-30:
Closing this issue as it has gone stale. Feel free to reopen if not.

closed 2024-04-30 by StoneT2000