mani-skill/ManiSkill #160: RuntimeError: Cannot find a suitable rendering device

ponyme, 2023-10-18:
I am facing the following error message when running my code on an Aliyun cloud server:
[2023-10-18 10:39:42.086] [svulkan2] [error] GLFW error: X11: The DISPLAY environment variable is missing
[2023-10-18 10:39:42.086] [svulkan2] [warning] Continue without GLFW.
Process ForkServerProcess-5:
Traceback (most recent call last):
  File "/home/pai/envs/maniskill2/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/pai/envs/maniskill2/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/stable_baselines3/common/vec_env/subproc_vec_env.py", line 29, in _worker
    env = _patch_env(env_fn_wrapper.var())
  File "/tmp/ipykernel_697/3208525770.py", line 16, in _init
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 801, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/mani_skill2/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/mani_skill2/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/mani_skill2/envs/pick_and_place/pick_cube.py", line 22, in __init__
    super().__init__(*args, **kwargs)
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/mani_skill2/envs/pick_and_place/base_env.py", line 27, in __init__
    super().__init__(*args, **kwargs)
  File "/home/pai/envs/maniskill2/lib/python3.10/site-packages/mani_skill2/envs/sapien_env.py", line 107, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: Cannot find a suitable rendering device
Output exceeds the size limit. Open the full output data in a text editor
---------------------------------------------------------------------------
ConnectionResetError                      Traceback (most recent call last)
Cell In[9], line 30
     27     return _init
     29 # create one eval environment
---> 30 eval_env = SubprocVecEnv([make_env(env_id, record_dir="logs/videos") for i in range(1)])
     31 eval_env = VecMonitor(eval_env) # attach this so SB3 can log reward metrics
     32 eval_env.seed(0)

File /home/pai/envs/maniskill2/lib/python3.10/site-packages/stable_baselines3/common/vec_env/subproc_vec_env.py:120, in SubprocVecEnv.__init__(self, env_fns, start_method)
    117     work_remote.close()
    119 self.remotes[0].send(("get_spaces", None))
--> 120 observation_space, action_space = self.remotes[0].recv()
    122 super().__init__(len(env_fns), observation_space, action_space)

File /home/pai/envs/maniskill2/lib/python3.10/multiprocessing/connection.py:250, in _ConnectionBase.recv(self)
    248 self._check_closed()
    249 self._check_readable()
--> 250 buf = self._recv_bytes()
    251 return _ForkingPickler.loads(buf.getbuffer())

File /home/pai/envs/maniskill2/lib/python3.10/multiprocessing/connection.py:414, in Connection._recv_bytes(self, maxsize)
    413 def _recv_bytes(self, maxsize=None):
--> 414     buf = self._recv(4)
    415     size, = struct.unpack("!i", buf.getvalue())
...
--> 379     chunk = read(handle, remaining)
    380     n = len(chunk)
    381     if n == 0:

ConnectionResetError: [Errno 104] Connection reset by peer

I have confirmed that 3 JSON files exist.

xuanlinli17, 2023-10-18:
Could you post your gpu type and your cuda version and your nvidia driver version?

ponyme, 2023-10-18:
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.82.01    Driver Version: 470.82.01    CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA A10          Off  | 00000000:00:08.0 Off |                    0 |
|  0%   31C    P8     8W / 150W |      0MiB / 22731MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

ponyme, 2023-10-18:
and `vulkaninfo` returned
ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object file: No such file or directory
'DISPLAY' environment variable not set... skipping surface info
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.

xuanlinli17, 2023-10-18:
Could you try the latest nvidia-driver (530+)?

Also what's your /usr/share/vulkan/icd.d/nvidia_icd.json

ponyme, 2023-10-18:
The latest verision of nvidia-driver compatible with cuda toolkit11.4 is 470, released on 26.6.2023.
And I do have libGLX_nvidia.so.0 in the library_path of the file

xuanlinli17, 2023-10-18:
Does the actual `libGLX_nvidia.so.0 ` file exist in the system?

ponyme, 2023-10-18:
I can't find the file in /usr/lib/x86_64-linux-gnu

xuanlinli17, 2023-10-18:
I think that's the issue. You need the actual file to exist in the system. If files similar to libGLX exists, please create a symlink.

ponyme, 2023-10-18:
yes, libGLX.so.0 exists and what should I do ?

xuanlinli17, 2023-10-18:
you need `libGLX_nvidia` to exist in the file name, since it's differerent from `libGLX.so.0`. What does `find / -name "*libGLX_nvidia*` return?

xuanlinli17, 2023-10-18:
If it doesn't exist I'd suggest to

sudo apt-get install libvulkan1
and reinstall nvidia-driver.

ponyme, 2023-10-18:
thanks for your reply and I'll give it a try.

StoneT2000, 2024-02-02:
Closing the issue as it is stale now.

closed 2024-02-02 by StoneT2000

HXY-95, 2024-03-01:
> Does the actual `libGLX_nvidia.so.0 ` file exist in the system?

On RTx 2080 and RTx 3090,  sapien works well. However, on A100, it does not work. All of these system do not have this file.

LovingPastry, 2025-06-05:
I meet the same problem, but I have check all details, and it doesn't work.
"""
$ sudo ls /usr/lib/x86_64-linux-gnu | grep libGLX_nvidia
libGLX_nvidia.so.0
libGLX_nvidia.so.570.133.07

$ nvidia-smi
Thu Jun  5 13:45:33 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.133.07             Driver Version: 570.133.07     CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 2080 Ti     Off |   00000000:01:00.0  On |                  N/A |
|  0%   40C    P8             11W /  300W |     733MiB /  11264MiB |      6%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
"""

StoneT2000, 2025-06-05:
You are missing a json file, see the troubleshooting guide here first: https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#ubuntu