mani-skill/ManiSkill #162: How can one use the differentiability of physics in soft-body environments?

pauljansonnie, 2023-10-19:
Hi,

Thanks a lot for providing these great environments.

Several existing elasto-plastic simulators/environments (PlasticineLab, DaXBench) are designed in such a way that gradient-based trajectory optimization can be easily leveraged to learn optimal trajectories w.r.t. some cost.
For ManiSkill2, I could not find examples that achieve this. The current code seems to hard-code that gradient should not be tracked by Warp. The variable `requires_grad` is implicitly set to `False` in all environments, e.g. in 
https://github.com/haosulab/ManiSkill2/blob/6f5b7afe2b16c18a82e3a064fd2f8fbf72cfc05e/mani_skill2/envs/mpm/pinch_env.py#L56-L58

Is there a particular reason why you made the decision of disabling differentiability of the physics? Do you see any obstacle that would make it difficult to implement this feature in the current codebase?

I have a bit of a related question but I can open another issue if you prefer: why did you decide to implement this elasto-plastic simulation using Warp? As other simulators rely on Taichi or JAX, I am curious to know your main motivations behind choosing Warp. Is there any particular reason apart from what is already stated in the ManiSkill2 paper (e.g. caching capabilities of Warp for faster startup)?

Thanks!

fbxiang, 2023-10-19:
While it is easy to add differentiability to the MPM framework, our system couples it with PhysX rigid body simulation, which is not differentiable. Implementing a fully differentiable rigid-mpm system is highly non-trivial so our whole system is not differentiable.

The main reason to use Warp is that it is easy to extend with CUDA and C++. Therefore it allows us to do any optimization (e.g. caching).

pauljansonnie, 2023-10-20:
Thank you very much for the prompt reply!

closed 2023-10-20 by pauljansonnie