mani-skill/ManiSkill #163: How to create a new envrionment for scratch.

saprrow, 2023-10-25:
Your work is really impressive and lots of sample envs are provided. And I'd like to know how to build **a custom envrionment**, for example a env where **several cups** filled with liquid on a desk. 

Thanks!

xuanlinli17, 2023-10-25:
Hi, thanks for your interest in our work! please refer to https://github.com/haosulab/ManiSkill2/blob/main/examples/tutorials/customize_environments.ipynb  for examples.

StoneT2000, 2023-10-26:
We will be releasing some better template code in the future as well as part of the next maniskill, but for now you can use the current tutorials

saprrow, 2023-10-27:
> We will be releasing some better template code in the future as well as part of the next maniskill, but for now you can use the current tutorials

Thanks for your reply. I'm looking forward to the futrue release🤗

saprrow, 2023-10-27:
> Hi, thanks for your interest in our work! please refer to https://github.com/haosulab/ManiSkill2/blob/main/examples/tutorials/customize_environments.ipynb for examples.

Thanks for your reply🤗

closed 2023-10-27 by saprrow